moggieuk/Happy-Hare #5: bug Missing [manual_mh_stepper selector_stepper] section in mmu_hardware.cfg

danorder, 2023-07-31:
current files don't seem to be able to find the [manual_mh_stepper selector_stepper] header filled in or clean off github.

moggieuk, 2023-08-03:
Can you give me some more details?  Does Klipper say "manual_mh_stepper" is an unknown section?  If so then it would be because the installer hasn't been run and all the Klipper "extra" python modules have not been installed.

danorder, 2023-08-05:
Sure using the installer the required files appear to load on under extras / uninstall the ercf version.   everything goes accordingly up until it claims the header isn't there. this would be be using your included file and, using the mellow ercf board pins. (i think its the same as the fystec)  and, for the sake of assurance yes the printer.cfg had the includes.. so im not certain why its having issues with that header. the file is seen since i can get errors on other settings before it reaches that point.  i can provide the log if required but the error i seen was essentially the same..

Fragmon, 2023-08-09:
i have the same with the combination with the klippain config.
https://github.com/Frix-x/klippain/tree/new-ercf_HHv2

Frix-x, 2023-08-09:
@benoitone since it's your work in Klippain, can you have a look if something is wrong ?
Is the branch in a working state ?

Benoitone, 2023-08-09:
> @Benoitone since it's your work in Klippain, can you have a look if something is wrong ? Is the branch in a working state ?

The branch is in working state for me... but only for the LGX Heavy extruder... for others extruders maybe thing about a different way to define them in klippain?

@Fragmon can you open an issue in Klippain repo for this? and post your printer.cfg?

moggieuk, 2023-08-23:
Thanks. So can I close this issue? Or is there something I should look at?  (sorry, still a little confused)

Benoitone, 2023-08-23:
Hello, yes the problem is with the include in printer.cfg. In klippain the mcu definition and Happy Hare files are already includes. So with klippain at the end of HH install script you must choose to do not include all the MMU config files.

moggieuk, 2023-08-26:
I can do that.  Can you give me a reliable thing to check for?  I.e. can you send your printer.cfg (or part of it) so I can see what to look for...

Benoitone, 2023-08-27:
Maybe use this line to do not include the config files by your install script:

### Klippain is designed to be used with the MMU/ERCF Happy Hare software: https://github.com/moggieuk/Happy-Hare

moggieuk, 2023-08-28:
Perfect. Thanks.

moggieuk, 2023-08-30:
Ok. Detection of existing Klippain installation is now in the installer so config files will not be automatically inserted.  Closing this issue.

closed 2023-08-30 by moggieuk

Frix-x, 2023-08-30:
Oh this is pretty cool ! Thanks for this addition to your install script :)