moggieuk/Happy-Hare #9: Endless Spool improvement / keep on pushing new Filament instead of unloading the old Filament

FlorianWilk, 2023-08-07:
Currently the EndlessSpool-Mode unloads the old filament when a clog is detected, selects the new gate and loads the new Filament.
So <bowden-length> Filament is always wasted.

Instead of unloading the old filament, ERCF could select the new Gate, load the new Filament and push it right after the old filament.
Then the old filament would be pulled from the extruder and the new filament would be pushed by the ERCF. 
Once the new filament has passed the Encoder and is directly after the old one (with no gap) we can just continue printing and no filament needs to be wasted.

This would require that there is still <bowden-length> of filament to print. If the print gets finished before the new filament reaches the extruder, the old filament would have no chance to get back to the ERCF and would get stuck in the Bowden.
In this case, unloading, it as it does now, would be neccessary.

Also this requires that sync-to-extruder is turned on. But this could be temporarily turned on until the new Filament reaches the Extruder. 

I currently do this by hand and it works pretty well - thanks to the sync-to-extruder feature. 
But an automatic solution would be great.

Btw: The manual way of doing this is also a pretty good way to calibrate/sync the rotation_distance of extruder- and ERCF-extruder. Just leave 1mm distance between the old and the new filament and check that it's still a 1mm gap when both ends reach the Extruder.

Surion79, 2023-08-15:
How does the encoder know when the new filament has reached the extruder? the previous filament is still being pulled and for unstopped printing, it would need to be right behind the old one, without any gap. since it is synced between extruder and gear motor, the encoder would not get any hint that something has changed. 
I wonder what the reliability of this feature is compared to a boden length waste. 
I wouldn't risk a 300 change print, if the endless spool is not reliable. In that case I would rather get 100g of waste on a spool instead of a boden length of it to be sure that the print doesn't fail because of it

FlorianWilk, 2023-08-15:
You would load it while the extruder is not printing/pulling on the "old" filament. When the encoder recognized, that the new filament is right behind the old one we could continue printing.

moggieuk, 2023-08-23:
I think this is an interesting idea.  I do see some challenges in the automation of this.  E.g. whilst the theory sounds fine I just experimented with ramming a new filament which has a pointed tip up to an old filament and it wedged quite tightly in my 2.5mm ID bowden tubes.    I think it is doable so will keep this "issue" open as a way to track.   I need to get some airtime with HHv2 first before embarking on this enhancement though.

moggieuk, 2023-09-04:
Thinking about this a little more.  What happens when a tool change is requested whilst a piece of filament is in the bowden?

FlorianWilk, 2023-09-04:
Hm, indeed. i was not thinking about multifilament prints. I am using die ERCF/HappyHare for initial filament selection for my prints most of the time. 
Just a thought: If happy hare had a internal purge mechanism (f.ex. something like a PelletPurge) we could purge it, because we know it's there. 
But you're right, it's challenging...

thisiscam, 2023-09-11:
> Thinking about this a little more. What happens when a tool change is requested whilst a piece of filament is in the bowden?

One solution is to lookahead the gcode file and determine if within a safety horizon we don't see any tool changes.

FlorianWilk, 2023-09-11:
> > Thinking about this a little more. What happens when a tool change is requested whilst a piece of filament is in the bowden?
> 
> One solution is to lookahead the gcode file and determine if within a safety horizon we don't see any tool changes.

I am not sure if HH has this information (the complete gcode), but indeed, this would be very smart. Also this would be a smart solution for the problem "what happens if the print is finished before the new filament arrives in the extruder"

Surion79, 2023-09-11:
> > Thinking about this a little more. What happens when a tool change is requested whilst a piece of filament is in the bowden?
> 
> One solution is to lookahead the gcode file and determine if within a safety horizon we don't see any tool changes.

Is this really possible? I thought that the gcode file is just being processed step by step without "look ahead". 
Even if so, would that not mean complex calculation when exactly it runs out? Feels for me like a lot of work for a "small" enhancement on saving 1-2m of filament per spool.

moggieuk, 2024-07-10:
I'm going to close this out.  I'm open to quality PR's, but this is incredibly difficult to do and although sounds like a good idea only really is useful for single color endless-spool prints.

closed 2024-07-10 by moggieuk