moggieuk/Happy-Hare #13: bug: Resume after failure pause incorrectly sets target temperature

bakatrouble, 2023-08-25:
When resuming print after failure pause, sometimes Klipper locks up, making me to reboot the firmware, resulting in an unfinished print

Log lines: 
```
- DEBUG: MMU_RESUME wrapper called
- - TRACE: Determined print status as: paused from print_stats
idle_timeout: Timeout set to 1800.00 s
- - TRACE: Determined print status as: paused from print_stats
- - TRACE: Determined print status as: paused from print_stats
- - TRACE: Determined print status as: paused from print_stats
Target temperature explicitly specified (202.9). Ignoring slicer
Target temperature explicitly specified (202.9). Ignoring slicer
- DEBUG: Waiting for extruder to reach target temperature (0.0)
```

I did not change the temperature manually, but the filament loading routine has waited for extruder to reach 200 degrees at some point during the pause.

bakatrouble, 2023-08-25:
Possibly the issue is in my custom PAUSE gcode which disables the extruder heater

bakatrouble, 2023-08-25:
Maybe it's not RESUME that hangs up

Quick reproduction steps:

- Start print w/o filament in the gate that will be loaded in the PRINT_START macro using `MMU_CHANGE_TOOL TOOL={TOOL} STANDALONE=1` command, Klipper pauses
- Wait for extruder to cool down below extrude temperature
- Load filament (i'm doing that using KlipperScreen, the command is `MMU_LOAD TEST=0`)

Then after the filament is brought to the extruder, the message `Deferring to slicer for extruder temperature (0.0)` is printed to console and nothing else happens

bakatrouble, 2023-08-26:
A possible fix might be forcing `self.paused_extruder_temp` as temperature override for `_ensure_safe_extruder_temperature()` calls if `self.is_paused_locked` is set?

moggieuk, 2023-08-26:
Thanks for the feedback.  I will take a deeper look...

bakatrouble, 2023-08-26:
Just tested my fix: now it correctly heats back up to the temperature on extruder loading

moggieuk, 2023-08-26:
Yes I can see the problem... (it was a side effect of removing the MMU_UNLOCK function -- which I still think was the right thing to do).   During a pause / resume interval anything that requires the extruder to move should reheat the nozzle...

I want to double check that you have all possible places caught but otherwise I'll merge - thanks.

During review I also noticed that I forgot to zero out the pause saved temperature on a cancel_print...

closed 2023-08-26 by moggieuk

moggieuk, 2023-08-27:
Ok.  I did incorporate your PR but then restructured the extruder temp setting logic quite a bit to make it easier to read.  If possible it would be great for you to update and try your test one more time to check that I haven't broken it again...

bakatrouble, 2023-09-01:
The same scenario (let extruder cool down during pause, then load filament) works fine

PR #14 (2023-08-26): Fix lockup if target temperature is reset during pause
Replicate V3 behavior: unlock before loading/unloading extruder if the print is currently paused
Fixes #13