moggieuk/Happy-Hare #19: Documentation: Explainer of terms in HH

PatrikTheDev, 2023-08-31:
It would be great to have a document (or a section in one of the existing) to explain all terms used within Happy-Hare, ERCF, MMU and/or Klipper. For example, the various options for homing are pretty unclear to someone just getting into MMU:

### Selector
- Endstop based (standard)
- StallGuard (maybe even explain the limitations of ERB/Easy-BRD regarding StallGuard setup on both motors at once as it is explained in the files)

### Gear motor
- Collision
- StallGuard
- Endstop? not sure if that's an option yet

### Extruder
- StallGuard (you call it extruder touch and nozzle touch in different places, it'd be better to keep it consistent)

IMO there should be an explainer to what everything means and what it actually does in practice

moggieuk, 2023-09-02:
This is a good point but I think a lot of what you describe should be (and will be) in the ERCF v2.0 manual.

But I'm totally open to a PR that I can easily integrate.  At the moment I'm burned out on doc writing...

moggieuk, 2023-12-18:
I'm going to close this because manual/doc has extensively been updated and this is not very actionable.  I'm happy to accept specific changes.

closed 2023-12-18 by moggieuk