moggieuk/Happy-Hare #20: Installation error

real-artswan, 2023-08-31:
What I've got running installer

`./install.sh -c ~/config -k ~/klipper`

```
Klipper config directory (/home/fly/config) found
Klipper service found
Reading default configuration parameters...
No previous mmu_parameters.cfg found. Will install default
No previous mmu_software.cfg found. Will install default
Copying configuration files into /home/fly/config/mmu directory...
expr: non-integer argument
./install.sh: line 415: ((: i<=: syntax error: operand expected (error token is "<=")
Linking mmu extension to Klipper...
Adding update manager to moonraker.conf
Restarting Moonraker...
Restarting Klipper...

Done.  Enjoy ERCF (and thank you Ette for a wonderful design)...

(\_/)
( *,*)
(")_(") MMU Ready

```

then I got nothing in all configs, like
[image]

moggieuk, 2023-09-02:
The error is annoying (and fixed now) but unless you specify "-i" you won't go through the interactive installer and thus the token placeholders will not be expanded.  In this scenario  (not using -i) you are expected to fill these in yourself.

Generally it is better to use "-i" the first time you install, then future updates can be done without the "-i"

real-artswan, 2023-09-02:
Thanks for the attention, i've used -i, strange... may be not...? I'll check tomorrow

real-artswan, 2023-09-08:
Thanks, installation works with current version

closed 2023-09-08 by real-artswan