moggieuk/Happy-Hare #26: ERCF v2.0 project nowhere to be found and "notes on Encoder problems" link on installation warning not working

birbou, 2023-09-03:
Basically what the title says, I wasn't able to locate anywhere the ERCF v2.0 project and when clicking on "[notes on Encoder problems](https://github.com/moggieuk/doc/ercf_encoder_v11.md) ", Installation section, I am getting "Not Found".

birbou, 2023-09-03:
Found out that the ercf v2 is not yet released but I'll leave this open because of the link not working

moggieuk, 2023-09-04:
I fixed the link to encoder notes (if using the original encoder rather than than the new Binky design).

Yes, ERCF v2.0 (community edition) is a project that I'm involved with but entirely separate from Happy Hare v2 which is now aiming to support all types of MMU.

matthewj301, 2023-09-09:
> Can you link to the ERCF v2 repository and/or mention it's availability? I've also searched for it and found nothing.
> 
> Does anyone know the release date? I'm about to build an ERCF v1.1 this week, so if I could build the v2 instead, that would make more sense...

I'd probably hold off, everyone in the community (Discord) is saying it should be any time now (mostly documentation cleanup now), or at least build your 1.1 with MoggieUK's Springy selector, the new [Binky Encoder](https://www.etsy.com/listing/1526104169/ercf-binky?ga_order=most_relevant&ga_search_type=all&ga_view_type=gallery&ga_search_query=binky+encoder&ref=sr_gallery-1-1&referrer_page_guid=91dbbf4d-898d-429b-9546-0fbd679e8f12&organic_search_click=1), and Happy Hare. 

ERCF V2.0 is going to bring some pretty large structural changes by incorporating (at least) the following mods:
- [Sturdy Bunny](https://github.com/sneakytreesnake/SturdyBunnyProject)
- [Springy](https://github.com/moggieuk/ERCF-Springy/tree/main) (mentioned above)
- [Ejection Assisted Purge](https://github.com/bombela/EjectionAssistedPurge)
- New [Binky Encoder](https://github.com/mneuhaus/EnragedRabbitProject/tree/main/usermods/Binky)

Cyberwizzard, 2023-09-10:
I think I'll have to wait then; I sincerely hope the redesign is beefier as I keep reprinting parts after destroying them. The latest being the gearbox housing holding the M5 nuts: when putting on the nylon nut at the end, I got to one whole turn before the gearbox part cracked as the Fystec provided nuts seem to require more torque than what the plastic can counteract...

Where the Voron was a delight to put together, the ERCF is sparking less joy...

vertigo235, 2023-09-12:
Got it, I am not sure why I missed it but the last time I checked the Voron discord, I thought the ERCF channel was gone. I must have looked in the wrong server :D

closed 2023-12-18 by moggieuk