moggieuk/Happy-Hare #27: _home_selector (MMU_HOME) ignores filament block width

garyd9, 2023-09-04:
_home_selector() appears to use the following formula to determine that max allowed movement:

`selector_length = 10. + (num_channels-1)*21. + ((num_channels-1)//3)*5. + (self.bypass_offset > 0)`

This appears to ignore the mmu_version string settings (and resulting cad_gate_width variable), instead using a hard-coded gate width of 21mm.  The end of the formula is also confusing to me, seemingly only adding 1 or 0 if bypass_offset is >0.

moggieuk, 2023-09-05:
Thanks. This was old logic from ERCF-Software-V3.  Fixing...

garyd9, 2023-09-06:
Not sure it's fixed, but unable to test.  Please see the (long) comment in #enraged_rabbit_open on the voron discord: https://discord.com/channels/460117602945990666/842792223769624605/1148780137257652286

garyd9, 2023-09-06:
Confirmed that the fix fails:
`08:54:47 Selecting filament bypass...
08:54:47 - DEBUG: Setting servo to move (filament hold) position at angle: 140
08:54:47 - - TRACE: Determined print status as: standby from print_stats
08:54:50 - - TRACE: Selector moved 124.5mm of intended travel from: 2.4mm to: 127.2mm (delta: 0.3mm)
08:54:50 - - TRACE: Truing selector 0.3mm to 127.2mm
08:54:50 - - TRACE: Setting MMU gear motor rotation distance ratio to 1.000000
08:54:50 Bypass enabled
08:54:56 Homing MMU...
08:54:56 - DEBUG: Moving up to 124.2mm to home a 6 channel MMU
08:54:56 - - TRACE: Determined print status as: standby from print_stats
08:54:58 - - TRACE: Determined print status as: standby from print_stats
08:54:58 - - TRACE: Determined print status as: standby from print_stats
08:54:58 Attempting to recover filament position...
08:54:58 - DEBUG: Checking for filament in encoder...
08:54:58 - DEBUG: Setting servo to down (filament drive) position at angle: 30
08:55:00 - - TRACE: After buzzing gear motor, encoder moved 0.00
08:55:00 - DEBUG: Filament not detected in encoder after buzzing gear motor
08:55:00 MMU [T?] >.. [En] ....... [Ex] ... [Nz] UNLOADED (@1052.2 mm)
08:55:00 - DEBUG: Setting servo to move (filament hold) position at angle: 140
08:55:00 An issue with the MMU has been detected whilst out of a print
         Reason: Homing selector failed because of blockage or malfunction. Klipper reports: No trigger on manual_mh_stepper selector_stepper after full movement`

garyd9, 2023-09-06:
Wow, I'm an idiot.  I just sent a PR to my own fork of this repo and it closed this issue.

garyd9, 2023-09-06:
If not obvious, I found the time to test the previous fix, and test an additional change that would resolve the issue.

closed 2023-09-09 by moggieuk