moggieuk/Happy-Hare #32: [Feature Proposal] Track gate statistic with exponential discounting of the past

thisiscam, 2023-09-06:
This gate statistic feature that came from v1 is nice. 
But, my OCD complains so much for the "Marginal" gates. So I noticed that some times it's just an occasional friction problem or an encoder misread that causes the MMU to think that a gate slipped. 
It seems like the current code is just accumulating all the slippage and compare that against the loaded/unloaded distance. 

IMO, occasional slippage should be tolerated if they are recoverable, and a gate only degrades if it keeps slipping. 
Therefore, I think this would be a nice to do some kind of exponential averaging of the slippage, so that the effect of slippage farther away in the past have diminishing effect on the present statistics.

moggieuk, 2023-09-09:
The "gate stats" were really envisioned to be able to help pinpoint problems.  I understand the idea to "average down" over time (which I will think about) but that would mess with the actual data which is reported at the DEBUG level and visible in mmu.log.    You can reset stats with various commands (e.g. MMU_STATS) to zero out the gate stats putting them all back into a "great" starting point.

I have a update in the works to improve on statistic keeping with both a historical and "last print" swap stats and more ways to clear just a portion of stats.

thisiscam, 2023-09-09:
> The "gate stats" were really envisioned to be able to help pinpoint problems. I understand the idea to "average down" over time (which I will think about) but that would mess with the actual data [...]

Yeah, I understand these points. Perhaps could record both averaged down and raw data (or just store the raw data, and compute the averaged down upon request lazily)?

moggieuk, 2023-09-10:
Yeah. Let me think about this because the update I have for per-print stats is only swap states, the gate states are kept until explicitly reset. 

Maybe I need a an extra stat for the averaged down and use that for the "great ... terrible" label.

moggieuk, 2024-02-05:
Implemented.

closed 2024-02-05 by moggieuk