moggieuk/Happy-Hare #36: Resuming after Pause during print changes temperature to MMU extruder default temp

matthewj301, 2023-09-09:
Running ERCF 1.11sb.

 When my printer resumes after either a manual PAUSE, or resuming after the printer is paused in order to initiate a tool change, ERCF sets my extruder temp to its self.default_extruder_temp (200).

My Klipper extruder min temp is set to 185. 

These are the logs showing the behavior.

- - TRACE: Synchronously loading filament to nozzle. Stepper: 'gear+extruder' moved 85.0mm, encoder measured 83.6mm (delta 1.4mm)
- DEBUG: Total measured movement: 83.6mm, total delta: 1.4mm
MMU [T0] >>> [En] >>>>>>> [Ex] >>> [Nz] LOADED (@763.5 mm)
MMU [T0] >>> [En] >>>>>>> [Ex] >>> [Nz] LOADED (@763.5 mm)
MMU load successful
MMU load successful
Loaded 752.9mm of filament (encoder measured 763.5mm)
Loaded 752.9mm of filament (encoder measured 763.5mm)
- DEBUG: MMU_RESUME wrapper called
- - TRACE: Determined print status as: printing from print_stats
Waiting for extruder to reach target temperature (200.0)
Waiting for extruder to reach target temperature (200.0)

Let me know what other info you need!

matthewj301, 2023-09-09:
Wondering if my toolchange macros are causing issues.

Not completely familiar with the codebase, but I see on "_MMU_RESUME" wrapper, you call "_ensure_safe_extruder_temperature" (here: https://github.com/moggieuk/Happy-Hare/blob/6416e02201701c6b8e63a18a32d9f046ce2e16a5/extras/mmu.py#L3879) and passing "paused_extruder_temp" class attr. 

you're getting the new_target_temp = self.paused_extruder_temp when printer is paused (here: https://github.com/moggieuk/Happy-Hare/blob/6416e02201701c6b8e63a18a32d9f046ce2e16a5/extras/mmu.py#L2036C6-L2036C6)

So I'm wondering if, my PAUSE macro doesn't go through, it doesn't know what the paused_temp is, and defaults to its known default_extruder_temp of 200? 

My macros below: 

[delayed_gcode RESUME_AFTER_TOOLCHANGE]
gcode:
  {% if printer.pause_resume.is_paused %}
    RESUME
  {% else %}
    CLEAR_PAUSE
    RESTORE_GCODE_STATE NAME=ercf_filament_change MOVE=1 MOVE_SPEED={printer["gcode_macro RatOS"].macro_travel_speed|float}
  {% endif %}

[gcode_macro CHANGE_ERCF_TOOL]
gcode:
  {% set ERCF_TOOL = params.NEXT_TOOL %}
  MMU_CHANGE_TOOL TOOL={ERCF_TOOL}
  UPDATE_DELAYED_GCODE ID=RESUME_AFTER_TOOLCHANGE DURATION=1

[gcode_macro PARK_AND_CHANGE_TOOL]
description: Park, change tool, and unpark
gcode:
  {% set ERCF_TOOL = params.NEXT_TOOL %}
  SAVE_GCODE_STATE NAME=ercf_filament_change
  PAUSE
  CHANGE_ERCF_TOOL NEXT_TOOL={ERCF_TOOL}

Also if you have advice on better macros for changing tools, please let me know :)

matthewj301, 2023-09-09:
Interesting, seeing that you actually pass that paused_extruder_temp to target_temp_override - so wouldn't even be evaluating the rest of that if-else if that is the case. If it is, not sure where the 200 comes from, other than here

https://github.com/moggieuk/Happy-Hare/blob/6416e02201701c6b8e63a18a32d9f046ce2e16a5/extras/mmu.py#L2049C9-L2055C31

matthewj301, 2023-09-09:
As a side note on debugging, it may be helpful to log the 'source' [here](https://github.com/moggieuk/Happy-Hare/blob/6416e02201701c6b8e63a18a32d9f046ce2e16a5/extras/mmu.py#L2070) as well, like you do here
                self._log_error("Heating extruder to **%s** temp (%.1f)" % (**source**, new_target_temp))

That way there's no question where the temp is coming from

matthewj301, 2023-09-09:
- - TRACE: Synchronously loading filament to nozzle. Stepper: 'gear+extruder' moved 85.0mm, encoder measured 84.6mm (delta 0.4mm)
- DEBUG: Total measured movement: 84.6mm, total delta: 0.4mm
MMU [T0] >>> [En] >>>>>>> [Ex] >>> [Nz] LOADED (@764.6 mm)
MMU [T0] >>> [En] >>>>>>> [Ex] >>> [Nz] LOADED (@764.6 mm)
MMU load successful
MMU load successful
Loaded 752.9mm of filament (encoder measured 764.6mm)
Loaded 752.9mm of filament (encoder measured 764.6mm)
- DEBUG: MMU_RESUME wrapper called
- - TRACE: Determined print status as: printing from print_stats
Ensuring Safe Extruder Temperature...
Ensuring Safe Extruder Temperature...
Current Vars - extruder: <kinematics.extruder.PrinterExtruder object at 0x7f98a5fd30> current_temp: 265.6 current_target_temp: 265.0 klipper min temp: 185.0
Current Vars - extruder: <kinematics.extruder.PrinterExtruder object at 0x7f98a5fd30> current_temp: 265.6 current_target_temp: 265.0 klipper min temp: 185.0
source: default, new_target_temp: 200.0
source: default, new_target_temp: 200.0
Waiting for extruder to reach target temperature (200.0)
Waiting for extruder to reach target temperature (200.0)

matthewj301, 2023-09-09:
I've gotten a good troubleshooting workflow causing it to do the above as its initially loading the extruder :)

This is possibly more interesting, Happy Hare is registering **is_in_print: True**, yet passes it in the if statement, and settles on **source: default, new_target_temp: 200.0**

- DEBUG: MMU_RESUME wrapper called
- - TRACE: Determined print status as: printing from print_stats
Ensuring Safe Extruder Temperature...
Ensuring Safe Extruder Temperature...
Current Vars - target_temp_override: 0.0 current_temp: 265.6 current_target_temp: 265.0 klipper_min_temp: 185.0
Current Vars - target_temp_override: 0.0 current_temp: 265.6 current_target_temp: 265.0 klipper_min_temp: 185.0
- - TRACE: Determined print status as: printing from print_stats
is_pause_locked: False **is_in_print: True** wait: True
is_pause_locked: False is_in_print: True wait: True
source: default, new_target_temp: 200.0
source: default, new_target_temp: 200.0
Waiting for extruder to reach target temperature (200.0)
Waiting for extruder to reach target temperature (200.0)

matthewj301, 2023-09-09:
Believe I found the symtom of the larger issue, target_temp_override is coming in as 0, evaluating to the first if statement (greater than -1), setting target temp to 0. This causes "if new_target_temp < klipper_minimum_temp" to evaluate to True, and setting new_target_temp to mmu's default temp

From debugging, right before the above evaluation:
Before target_temp < minimum_temp eval - source: resume new_target_temp: 0.0 klipper_minimum_temp: 185.0

matthewj301, 2023-09-09:
Will open a PR for that, I think best solution is to check that target_temp_override is greater than klipper_min_temp, rather than default of func var of -1

matthewj301, 2023-09-09:
Related to issue https://github.com/moggieuk/Happy-Hare/issues/37 and PR https://github.com/moggieuk/Happy-Hare/pull/38 it seems

moggieuk, 2023-09-09:
Thanks for all the research. There were changes in this area of code recently and clearly a bug crept it.   I'll look at your PR shortly.

Can you tell me more about the "speed" change?  Are you referring just to the movement back to the print (which results from essentially a RESTORE_GCODE_STATE call) or is your print speed reduced for rest of the print? If the latter, do you use M220 (or M221) when paused?  It is a known issue that this doesn't get restored after toolchange and a enhancement to fix that is in the works.

matthewj301, 2023-09-09:
No problem, and agreed it should be fixed at the more root cause, but I don't think comparing incoming target_temp_override against klipper_min_temp is a bad thing for general safety and clarity. But either way, I'm glad we got to the bottom of it :)

Regarding the speed issue, I tried just using T{tool} with the z hop height (very nice addition btw) for my toolchange (middle of print), so that Happy Hare did the complete state save and restore, and after it restores the gcode state and continues printing, the print speed is multiples below what it was prior to the tool change. I'm wondering if using z_hop_speed on restore_gcode_state is causing something to get messed up. I confirmed my velocity and acceleration all appeared normal after the toolchange, yet print speed was dropping down to 30mm/s (also my z hop speed)

matthewj301, 2023-09-09:
Regarding the official hooks (_MMU_PRE_UNLOAD/LOAD), I really want to use them, but i found that I needed different logic during loading at start of print, and in the middle of print, so to use these I would need to write more logic around the state of the printer. For me, I don't pause the printer when initially loading (during my start_print macro), but do when initiating tool change (so it parks away from print) 

It would be cool if there were built-in methods for the above, since Hare already understands printer state - load_on_start, load_during_print, etc

Even purging macros built in, one day :)

matthewj301, 2023-09-09:
I missed your question on M221, was on my phone - will check now.

matthewj301, 2023-09-09:
New PR that addresses self.paused_extruder_temp not being set when a PAUSE is manually issued. Let me know if I missed any edge cases.

Personally, I think my previous PR is good from a defensive programming perspective, but up to you!

Thanks again for all your work on this

moggieuk, 2023-09-10:
PR merged.  The paused temp will be saved in a standard PAUSE wrapper in the next major release, but this is a good solution until then.  THanks.

matthewj301, 2023-09-10:
Sounds good, looking forward to it!

closed 2023-09-10 by matthewj301