moggieuk/Happy-Hare #37: Endless Spool - not restoring temperature

garyd9, 2023-09-09:
My system is using the "client_macros.cfg" provided with HappyHare (v2) ([include mmu/optional/client_macros.cfg])

I created an endless spool group between 0 and 4, and started a print with the bed at 117C and the extruder at 255C (for all filaments.  The gcode being printed never changes the temperature after starting the print.)  I purposely cut the filament going into filament block 0 and waited to see what happened.

As expected, the runout was detected...
```
08:25:25 EndlessSpool checking for additional spools in group 1...
08:25:25 Remapping T0 to gate #4
08:25:28 Forming tip...
...
08:25:51 - DEBUG: Loading tool T0...
08:25:51 - DEBUG: Selecting tool T0 on gate #4...
```
However, something changed the extruder tool temperature to 200C (instead of leaving it at 255C) and nothing in the provided macros restored the original temperature.  This resulted in a print clog.
```
08:26:18 - DEBUG: MMU_RESUME wrapper called
08:26:18 - - TRACE: Determined print status as: paused from print_stats
08:26:18 Waiting for extruder to reach target temperature (200.0)
08:26:56 - DEBUG: Restoring toolhead position
```

It would be desirable if the default macros and tools would either not alter the extruder tool temperature whatsoever, or if it does, it restores it to the temperature at the time the runout was detected.

garyd9, 2023-09-09:
I'm not sure when HH is changing the extruder temperature.  However, assuming it's doing it _after_ _MMU_ENDLESS_SPOOL_PRE_UNLOAD is called, and _before_ _MMU_ENDLESS_SPOOL_POST_LOAD is called, a quick and dirty fix would be to save the extruder temp to a variable in _MMU_ENDLESS_SPOOL_PRE_UNLOAD and would be restored in _MMU_ENDLESS_SPOOL_POST_LOAD.  For example:

```
[gcode_macro _MMU_ENDLESS_SPOOL_PRE_UNLOAD]
description: Optional post unload routine for EndlessSpool changes
gcode:
    SET_GCODE_VARIABLE MACRO=_MMU_ENDLESS_SPOOL_POST_LOAD VARIABLE=extemp VALUE={printer['extruder'].target}
    PAUSE

[gcode_macro _MMU_ENDLESS_SPOOL_POST_LOAD]
description: Optional post load routine for EndlessSpool changes
variable_extemp: 0
gcode:
    {% if extemp > 0 %}
        M109 S{extemp|int}                                                        ; wait for hotend to heat back up
    {% endif %}
    RESUME
```

garyd9, 2023-09-09:
I've narrowed down, in klippy.log, where the temperature was changed.  (look for "extruder: target=" in the "Stats" lines. 
 It _may_ be within or immediately after "MMU_RESUME wrapper" that the target temp is changed.

```
Stats 169339.9: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000005 mcu_task_stddev=0.000004 bytes_write=8538496 bytes_read=22477754 bytes_retransmit=9 bytes_invalid=0 send_seq=358740 receive_seq=358740 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180000063 mmu: mcu_awake=0.009 mcu_task_avg=0.000016 mcu_task_stddev=0.000028 bytes_write=1498971 bytes_read=33820560 bytes_retransmit=0 bytes_invalid=0 send_seq=195380 receive_seq=195380 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=11999647 adj=11999637 EBBCan: mcu_awake=0.003 mcu_task_avg=0.000011 mcu_task_stddev=0.000011 bytes_write=3642040 bytes_read=20618725 bytes_retransmit=0 bytes_invalid=0 send_seq=229983 receive_seq=229983 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63998767 adj=63998742  SB2209: temp=72.7 heater_bed: target=117 temp=116.9 pwm=0.378 enclosure_temp: temp=52.0 sysload=0.79 cputime=5796.126 memavail=3419320 print_time=169349.864 buffer_time=0.752 print_stall=2 extruder: target=255 temp=255.2 pwm=0.282
- - TRACE: Synchronously loading filament to nozzle. Stepper: 'gear+extruder' moved 72.0mm, encoder measured 71.6mm (delta 0.4mm)
- DEBUG: Total measured movement: 71.6mm, total delta: 0.4mm
MMU [T0] >>> [En] >>>>>>> [Ex] >>> [Nz] LOADED (@1080.7 mm)
MMU [T0] >>> [En] >>>>>>> [Ex] >>> [Nz] LOADED (@1080.7 mm)
MMU load successful
MMU load successful
Loaded 1095.2mm of filament (encoder measured 1080.7mm)
Loaded 1095.2mm of filament (encoder measured 1080.7mm)
- DEBUG: MMU_RESUME wrapper called
- - TRACE: Determined print status as: paused from print_stats
Waiting for extruder to reach target temperature (200.0)
Waiting for extruder to reach target temperature (200.0)
Stats 169340.9: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000005 mcu_task_stddev=0.000004 bytes_write=8538670 bytes_read=22478136 bytes_retransmit=9 bytes_invalid=0 send_seq=358753 receive_seq=358753 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180000063 mmu: mcu_awake=0.009 mcu_task_avg=0.000016 mcu_task_stddev=0.000028 bytes_write=1499033 bytes_read=33820868 bytes_retransmit=0 bytes_invalid=0 send_seq=195385 receive_seq=195385 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=11999647 adj=11999637 EBBCan: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000014 bytes_write=3642074 bytes_read=20618918 bytes_retransmit=0 bytes_invalid=0 send_seq=229986 receive_seq=229986 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=8 upcoming_bytes=0 freq=63998761 adj=63998746  SB2209: temp=72.7 heater_bed: target=117 temp=116.9 pwm=0.378 enclosure_temp: temp=52.1 sysload=0.79 cputime=5796.192 memavail=3416988 print_time=169350.224 buffer_time=0.112 print_stall=2 extruder: target=200 temp=255.0 pwm=0.000
```

garyd9, 2023-09-09:
...and even more comments, because I'm incapable of encountering a bug and not pursuing it...

Working backwards:  

I see that the cmd_MMU_RESUME wrapper is called by the endless spool code.  This is reflected by the helpful log entry:
   DEBUG: MMU_RESUME wrapper called
  
Checking the code in cmd_MMU_RESUME, I noticed that it calls:
```
# Important to wait for stable temperature to resume exactly how we paused
self._ensure_safe_extruder_temperature(self.paused_extruder_temp, wait=True)```
```        
I believe that at this point in the code, self.paused_extruder_temp is 0.  Why?  Because the only place that paused_extruder_temp is set to non-zero is in "_pause()", which the endless spool code seem to ever call.

Then _ensure_safe_extruder_temperature() then realized that the passed temp is < self.default_extruder_temp, and therefore changes the target to self.default_extruder_temp.  Then it sets that new temperature (which is 200) and WAITS for the temperature to change to self.default_extruder_temp +/-1.

-------

All that being said, my suggested workaround is clumsy as it would result in the extruder going from 255, then waiting to cool it to 200, then waiting to heat it back up to 255.  

The easiest fix, I think, would be for _handle_runout() to simply set self.paused_extruder_temp to the current extruder temperature as soon as it recognizes that it has another gate to switch to. 

I'll submit a PR for that...

moggieuk, 2023-09-09:
Thanks for the analysis and fix.  PR merged

garyd9, 2023-09-09:
Tested and confirmed fixed.  Thank you.

closed 2023-09-09 by garyd9