moggieuk/Happy-Hare #40: Extruder heating sequence

lettore, 2023-09-09:
Coming from HH v1.1 and experiencing many weird situations with HH v2.
I see that in loading sequence the extruder is heated up only when it reach the extruder zone, before was done at the start of the sequence.
More unbelievable is the swapping sequence, it tries to unload the filament without heating, then load another filament (obviously the previous one is not moved from the extruder) then heat up the extruder when it is supposed to be at the extruder entry and finish the procedure correctly. 
While the original filament never moved from the extruder, it get grinded and the encoder never moved.
It's amazing that you reworked all the code to be compatible with other kind of units, but I should say, ERCF Software V3 was really rock solid for me, why an upgrade should be a step backward in reliability versus the old version?

moggieuk, 2023-09-10:
Thanks for the feedback and noted.  Will look into it.

Reason for new code is here: https://discord.com/channels/460117602945990666/909743915475816458/1146667229795078165

moggieuk, 2023-09-10:
BTW If you update you will probably find the heating problem fixed.

lettore, 2023-09-10:
Just to start, this is your work and I greatly appreciate it, otherwise, I will use the ERCF official version that I really don't like.
That said, you had a rock-solid product with v1.1, you added support for multiple MMU (btw the support is still wip) ruining the already working ERCF section. I can stick with v1.1 but, as Klipper Screen now is broken because v1.1 is no more supported, and you didn't make a tag to the most recent compatible version I'm forced to use v2.
So I'm ready to do the alpha testing and report the problem I find, just don't be mad if I'm still wondering why with the old version I had 0 problems in months and with an upgrade I'm unable to make a tool change.

thisiscam, 2023-09-11:
Hi! 

>  if I'm still wondering why with the old version I had 0 problems in months

It's normal for an open source project to be no longer maintained due to people having various other duties and obligations (in this case the main maintainer of v1.1 @moggieuk is spending most of his time working on v2, I'm guessing). For me the old 1.1 software worked rock solid as well --- if you feel that's the case, just don't upgrade. Or downgrade to whatever klipper/klipperscreen/HHv1 versions that you were happy with. 

> you didn't make a tag to the most recent compatible version I'm forced to use v2.

You can try to recall on what date your printer was still in the rock solid state.
Then do `git log` in `HHv1`, find the first commit before that date, then `git checkout {COMMIT}`.  Same is for klipper/klipperscreen

> with an upgrade I'm unable to make a tool change.

HHv2 is new software with new features, and it's normal for such new softwares to have bugs.

garyd9, 2023-09-11:
> ... just don't be mad if I'm still wondering why with the old version I had 0 problems in months and with an upgrade I'm unable to make a tool change.

Wow.  They shouldn't be mad if you come across as ungrateful, but you're allowed to be mad that the software that you didn't pay for and someone spent countless hours working on isn't perfect?  I understand that you're frustrated, but the author doesn't owe you anything.  

The approach you took can discourage people like moggieuk from contributing to the community.  That hurts everyone (including yourself.)

If it's not working for you, I'd suggest HELPING to solve the problems, instead of sounding ungrateful and demanding.  How can you help?  Easy:

1.  If you're familiar with python, you can find/fix issues and submit pull requests.

2.  Even if you aren't familiar with python, you can spend the time to triage the issue, find steps to repeat, and document as much as possible so that others understand what you think is a problem, how to repeat it, and what you expected to happen.

Above all, remember that all of this is something someone has spent an unimaginable amount of time working on, and then allowed you to use for free.  The proper response is "thank you."

lettore, 2023-09-12:
I'm not ungrateful, there's one big problem with open source,  it's that you're offering your work for free to people to use in real life. When you have hundreds of users that use your software you also have the responsibility to not broke the whole thing with an update. Everyone is using Klipper here, it's open source as well, but you don't have to worry if you update to last version you have to rework everything and you end up with a non working system. So mine was just an advice, if you have a large customer base not try to reinvent the wheel if you have an already rock solid product. I think moggieuk perfectly understood my point, and he's maybe the only one that can complain.

thisiscam, 2023-09-12:
> So mine was just an advice, if you have a large customer base 

I consider myself (and probably all of us) a user not a customer.

> not try to reinvent the wheel if you have an already rock solid product.

HHv2 supports MMUs other than ERCF, which some of the ERCF-V3 users (including myself) might or will find useful. 
As an analogy, the situation is more like "smartphone" (HHv2) vs "cellphone prior to 2007" (rock solid product / ERCF-V3).
It's true that the set of users of HHv2 is likely to overlap with the set of users of ERCF-V3. But, HHv2 and ERCF-V3 are two separate softwares. 

Also, quoting @moggieuk 's discord words on the release of HHv2:
```
5) Final, final, final word:  If you are happy with HHv1 (aka ERCF-Software-V3) then don't feel pressured to upgrade now. You can always wait and do so in the future after all the wrinkles are worked out.
```
^^ which is basically what I replied above --- try to revert to the old ERCF-V3. If you have trouble with that, I am willing to help.

lettore, 2023-09-12:
I opened this just to report a bug, the bug is now fixed for me (tried some tool change in various conditions and it worked well). All the opinions and suggestions I received after is well accepted but not related to the bug itself.

closed 2023-09-12 by lettore