moggieuk/Happy-Hare #43: [Feature Request] Load from buffer for the first time if not already during calibration

thisiscam, 2023-09-11:
I think we should do a preload from the buffer prior to doing any of the gate calibrations. Right now MMU is loading in slow speed for loading and fast speed for unloading during `MMU_CALIBRATE_GATES ALL=1`

garyd9, 2023-09-11:
Just my opinion:  

I tend to calibrate everything with loose filament cut off the spool.  I do this for 2 reasons:  

1. I don't want friction, buffer, and spool resistance impacting the calibration.  Remember that the calibration is meant to measure the actual movement relationship between the gear and encoder.  If the filament is slipping on the gear (as might happen even in a preloaded buffer because of other friction), the calibration would be wrong.

2. I also tend to use "fresh" filament for calibrating, so there's no gear impressions already pressed on the filament.  This, also, is to promote as much accuracy from the calibration as possible.

If the encoder isn't calibrated accurately, I wouldn't be able to trust the slippage values that HH reports, and the "bowden_apply_correction" setting would be worthless.

thisiscam, 2023-09-11:
> Just my opinion:
>
> I tend to calibrate everything with loose filament cut off the spool. I do this for 2 reasons:
>
> 1. I don't want friction, buffer, and spool resistance impacting the calibration. [...]

Not sure if I get your point, or if this issue was vague. The issue is about setting the speed during calibration. Right now HH seems to be using an inferred speed that might be either the buffer loading speed or the non-buffered loading speed.  I'm proposing to do a preload so that HH will then always use the fast buffer loading speed during calibration. I am guessing that you are implying that you want an another speed setting, just for the calibration (and users need to be responsible for making sure the filament is not in the buffer)?

garyd9, 2023-09-12:
> Not sure if I get your point, or if this issue was vague. The issue is about setting the speed during calibration. 

I'm confused.  If it's **only** about setting the speed, why use terms like "preload from the buffer"?  Did you mean to suggest something like "use the speed from 'gear_from_buffer_speed' in the parameters file"?

Even still, when calibrating, slower is better in my opinion.  Slower is more likely to be accurate (and the calibration only needs to be done once.)  Assuming a well assembled ERCF, an accurately calibrated HH install is a dream to use.  Rushing through the calibration might save you 2-3 minutes when you're first setting up, but cost you hours upon hours later.

thisiscam, 2023-09-12:
> I'm confused. If it's **only** about setting the speed, why use terms like "preload from the buffer"? Did you mean to suggest something like "use the speed from 'gear_from_buffer_speed' in the [...]

It's about how the current implementation sets the speed and that it's not ideal. I proposed one solution which is to preload, then always use this  `gear_from_buffer_speed `.  

> Even still, when calibrating, slower is better in my opinion. Slower is more likely to be accurate (and the calibration only needs to be done once.) 

I don't think that is correct. From my observation, loading with a lower speed tends to result in shorter length of loaded filament. This is likely due to 1. the friction is speed dependent 2. faster loading means greater momentum. 
Further, it is not clear what it means to be accurate. In particular, the bowden tube has a lot of slack. Thus, the calibration is not really measuring the length of the bowden tube, but the amount of filament that can fit into the bowden tube --- this amount is speed/friction dependent. 
Therefore, IMO, the calibration should be done in an environment as close to the deployment as possible --- including buffer and loading speed.

garyd9, 2023-09-12:
I respectably disagree with you.  

First you run "MMU_TEST_MOVE" and "MMU_CALIBRATE_GEAR" so HH learns EXACTLY how many steps to move the gear motor for a given length when the filament is in gate 0.

Then you run MMU_CALIBRATE_ENCODER so HH learns exactly how many pulses of the encoder correlate to an exact length of filament as measured by exactly how many steps the gear motor moves when the filament is in gate 0.

Finally, because the gears in gates 1 thru n might not be cut exactly like the gears in gate 0, MMU_CALIBRATE_GATES measures the difference of encoder pulses when the filament is in the other gates.  If the gears are identical, CALIBRATE_GATES should come back with a bunch of "1.0"'s.   (My gears are all identical, and since I calibrated with loose filament, all my gates calibrated to 1.0.)  At this point, it should be measuring with no friction and no momentum.  

The point of calibrating is so that the encoder can know how much filament passes by under perfect circumstances (or as perfect as can be obtained.)   Going faster (or even having the filament in a buffer) makes the circumstances less perfect, and results in a poorer calibration.

thisiscam, 2023-09-12:
> The point of calibrating is so that the encoder can know how much filament passes by under perfect circumstances (or as perfect as can be obtained.) 

Could you define what is the perfect the circumstance for you?

garyd9, 2023-09-12:
> Could you define what is the perfect circumstance for you?

as perfect as I'm able to obtain is:  Filament that isn't constrained and supported so it has a free flow into the filament block.  I also try to warm the filament a bit before calibrating to try and get the curl out of it (and let it cool back down, of course.)  (yeah, I'm kind of OCD.)

It's nothing at all like the circumstances of the filament that feeds in to the block in actual printing, but because I have a very accurate encoder calibration, the encoder notices those differences.

The idea isn't to get the mmu.log to report deltas of 0.  The idea is to get the encoder to see that the gear tried to move 100mm, but actually only moved 99mm.

thisiscam, 2023-09-12:
> as perfect as I'm able to obtain is: Filament that isn't constrained and supported so it has a free flow into the filament block. I also try to warm the filament a bit before calibrating to try and [...]

What is the loading speed for this circumstance? I am assuming that it is $\epsilon$-close to 0? 
Under this theoretical circumstance, say you try to run `MMU_CALIBRATE` and the encoder (which for our sake let's assume it's a perfect measuring device) says it measured a traveled distance of 900mm. 
But, if you run `MMU_CALIBRATE ` with a loading speed of 300mm/s, the encoder might measure a traveled distance of, say, 905mm. 

Which one should you believe? Well, in fact, both can be correct! The traveled distance is actually a function of loading speed. Think about it like this: when the filament moves at different speeds, it bends into different shapes within the bowden tube; so with different speeds you actually get different length of filament up to the extruder. 

As much as I like to believe there exists a perfect circumstance, there doesn't seem to exist one even theoretically. 

I might agree with calibrating the encoder with everything taken off. But for calibrating the gates, I would use the deployment setup.

garyd9, 2023-09-12:
> I might agree with calibrating the encoder with everything taken off. But for calibrating the gates, I would use the deployment setup.

MMU_CALIBRATE_GATES _is_ a calibration of the encoder.  You run "MMU_CALIBRATE_ENCODER" to calibrate gate0, and MMU_CALIBRATE_GATES to determine how gates 1 thru n differ from gate0.  

Think of it this way:  MMU_CALIBRATE_ENCODER generates a variable called "mmu_encoder_resolution."  That's the encoder calibration when the filament is in gate0.  When the filament is in gate3, it modifies mmu_encoder_resolution by the value generated by MMU_CALIBRATE_GATES (mmu_calibration_3 for gate3.)   (mmu_calibration_0 is always "1.0")

thisiscam, 2023-09-12:
> MMU_CALIBRATE_GATES _is_ a calibration of the encoder. You run "MMU_CALIBRATE_ENCODER" to calibrate gate0, and MMU_CALIBRATE_GATES to determine how gates 1 thru n differ from gate0.
>
> Think of it [...]

Ah I see where we talked pass each other. The old ERCF software had a command `ercf_calibrate` which calibrates the bowden tube length of gate 0 (and does the relative measurement for all the rest of the gates), which I thought is what now called `MMU_CALIBRATE_GATES ` (I was wrong)  

Given that, what I am proposing is to do a preload prior to `MMU_CALIBRATE_ENCODER ` that calibrates the bowden tube length.

For the relative calibration, you should either take everything off, or load from the buffer --- assuming your setup works fine, then there should already be minimal friction when loading from buffer.  In the later case, a preload also makes sense.

moggieuk, 2023-09-12:
You are correct that ALL calibration should be done without friction.  I recommend a loose piece of filament...  

I'm happy to see if it can be made clearer in the doc but I don't want to complicate the calibration procedure - although it is easy for experienced users, the first time person has a really hard time with it.

That said, it would be possible to simply to +1 pull/push sequence and discard the first result.  I haven't read all of this long thread but is that what you are suggesting?

thisiscam, 2023-09-12:
> You are correct that ALL calibration should be done without friction. I recommend a loose piece of filament...
>
> I'm happy to see if it can be made clearer in the doc but I don't want to complicate [...]

Yes that is what I am suggesting, except that this additional pull push should to be needed if the filament is already buffered.

moggieuk, 2023-09-28:
I added to my TODO list