moggieuk/Happy-Hare #45: Other board options request

mlee12382, 2023-09-13:
Would it be possible to add pins for the Mellow Easy BRD CAN board and the new BTT MMU board? I am including links for the pinouts.

https://mellow.klipper.cn/#/board/fly_ercf/pins

https://github.com/bigtreetech/MMB/blob/master/Hardware/MMB%20CAN%20V1.0-Pin.jpg

moggieuk, 2023-09-15:
Since I can't verify these I really need some help in the form of a PR for the install.sh script.  I will try to add more boards over time but can't guarantee when.

mlee12382, 2023-09-15:
I'm not sure how to integrate all the install prompts but here's my mmu.cfg for the Mellow CAN board: https://github.com/mlee12382/OctopusMaxEz-Backup/blob/master/mmu/base/mmu.cfg
Also I was trying to alias my toolhead board but it was giving me errors, I even tried adding the mcu: SB2209 line instead and that didn't work either. Do you have any insights? I ended up just bypassing the extruder aliases and leaving them manual entries to make it work.

Eisbaeeer, 2023-11-11:
Im using same board with CAN. I run a working setup in the past with this board and happy-hare in earlier verison. The older config attached. Now I reinstalled all things and don´t get it running, because the config changed and there are not the same namings. Perhaps my config helps to get the right settings. 
[ercf_hardware.zip](https://github.com/moggieuk/Happy-Hare/files/13326533/ercf_hardware.zip)

if you need more details, don´t hasitate to post here.

mlee12382, 2023-11-11:
> Im using same board with CAN. I run a working setup in the past with this board and happy-hare in earlier verison. The older config attached. Now I reinstalled all things and don´t get it running, [...]

Here's my mmu.cfg for the Mellow CAN board:
```
[mcu mmu]
canbus_uuid: f079bd60e64b # Config guess. Run ls -l /dev/serial/by-id and set manually	# Change to `canbus_uuid: 1234567890` for CANbus setups

[temperature_sensor MMU]
sensor_type: temperature_mcu
sensor_mcu: mmu

# Section to create alias for pins used by MMU for easier integration into Klippain and RatOS.
# The names match those referenced in the mmu_hardware.cfg file. If you get into difficulty
# you can also comment these alisas and configure the pin names directly into mmu_hardware.cfg.
#
[board_pins mmu]
mcu: mmu # Assuming using an external / extra mcu dedicated to MMU
aliases:
    MMU_GEAR_UART=gpio9,
    MMU_GEAR_STEP=gpio7,
    MMU_GEAR_DIR=gpio8,
    MMU_GEAR_ENABLE=gpio6,
    MMU_GEAR_DIAG=gpio23,
    MMU_SEL_UART=gpio0,
    MMU_SEL_STEP=gpio2,
    MMU_SEL_DIR=gpio1,
    MMU_SEL_ENABLE=gpio3,
    MMU_SEL_DIAG=gpio22,
    MMU_SEL_ENDSTOP=gpio20,
    MMU_SERVO=gpio21,
    MMU_ENCODER=gpio15
```
Did you uninstall Happy-Hare v1 before installing v2? Probably easier to start from scratch and I think Moggieuk mentioned somewhere that it works better if you uninstall the old version first.

Eisbaeeer, 2023-11-12:
Got it working with manual mode. The aliases didn´t worked for me. Perhaps this is reagarding the CAN namings of pin´s. The calibration routines are much easier then in earlier version. Thank´s for help. The issue can be closed.

mlee12382, 2023-11-12:
> Got it working with manual mode. The aliases didn´t worked for me. Perhaps this is reagarding the CAN namings of pin´s. The calibration routines are much easier then in earlier version. Thank´s for help. The issue can be closed.

I also had issues with aliases for the can board pins. Seems to be a common issue. I think the current version of HH doesn't need them anyway since you're no longer moving the extruder settings to HH cfg files.

Eisbaeeer, 2023-11-15:
It seems to fixed with version: v2.2.0-3-g3d41ecc7
I installed HH from the scratch and defined the alias. It worked like a charme.

moggieuk, 2024-02-05:
More boards are now supported in installer.

closed 2024-02-05 by moggieuk