moggieuk/Happy-Hare #46: Multiple loading or unloading attempts option.

DianosOdd, 2023-09-13:
Maybe I'm missing something in Happy Hare 2 but my most frequent error is a failure to feed the extruder and being detected by the toolhead sensor.

I know I can use the option to ignore those failures and just get the error and continue but a manual unloading, loading routine fix the issue and would be awesome to be automated in case of this failure.

Eventually if there is a way to compile a routine as a gCode to set a number of retries would be helpful to know.

Sry in advance for my awful English.

DianosOdd, 2023-09-15:
Thanks a lot I didn’t find it browsing. 
I totally see your point but until I keep swapping filaments forgetting to recalibrate extruders it mitigates my error.

moggieuk, 2023-09-17:
It's in the doc but to pull into your `mmu_parameters.cfg` just re-run: `./install.sh` in the Happy-Hare directory.

With toolhead sensor (and `toolhead_sync_load: 1`) you should be getting 99.99% reliability on extruder load...