moggieuk/Happy-Hare #48: [Feature Proposal] Use Encoder even ERCF is disabled

FOLeader, 2023-09-14:
When ERCF is disabled, all functions of ERCF are off.
It would be nice, if the encoder still works for clogs and runout as option

moggieuk, 2023-09-19:
This is actually support but not when ERCF is disabled.  Instead you leave it enabled and select the bypass.

`MMU_SELECT_BYPASS`

The bypass is an optional "mod" but is part of the upcoming ERCFv2.  You __could__ spoof it by reducing your number of gates by one and configuring the last as the bypass - it's not documented but with direct manipulation of `mmu_vars.cfg` it would be possible.

Another way is to use klippers `duplicate_pin_override`, configure the encoder as a completely separate filament_motion_sensor and enable only when ERCF is disabled -- never tried this but I think would be possible.

FOLeader, 2023-09-19:
thank you very much,

the last idea sounds, i will check it :)