moggieuk/Happy-Hare #49: Prusa MMU

Ronnywa09125, 2023-09-14:
Hello,
I am in the process of integrating a Prusa MMU2 clone that runs with Klipper. Have you already completed something for the Prusa and your Happy-Hare and your Klippercreen?

Greating Ronny

moggieuk, 2023-09-15:
Not yet. All the plumbing is there but is it not complete.  Just ERCF at the momement.

Ronnywa09125, 2023-09-16:
Ok, Thanks.

Lino77, 2023-10-16:
If you need someone to test it, let me know. My config is an MK3 with a BTT002 as MCU, and the MMU with the Skr E3

ErdbeermarmeladebrotmitHonig, 2023-12-20:
Hello,
i just want to know if it will be possible to keep the hardware of the Prusa MMU or do I have to get a new controller board?
I already read about flashing Klipper onto the small MCU on the Prusa MMU hardware is hard and requires to exclude some modules.

Pepeuch, 2023-12-30:
hello, I am currently installing my prusa mmu2s. I understood that it was still not supported by happy-harre, but I would still like to try the experiment because the only thing that is really different from ercf is the use of a third stepper (the iddler). Do you think it is possible to replace the servo in the code with this motor or do you have to rewrite everything?

moggieuk, 2024-02-05:
Sorry, I only just saw this old comment.  Rewriting the servo as a class to move stepper instead is work in progress for me.  Still not done but close -- but I'm taking exactly the same approach as you suggested.

moggieuk, 2026-10-21:
Thanks for asking, and sorry this sat open so long. To answer it directly: Happy Hare v4 (v4.1 is the current release) still doesn't have a built-in Prusa MMU2/MMU3 type. If you set the vendor to Prusa, it stops with a "not yet supported" error.

There is community work on running a Prusa MMU with Klipper and Happy Hare. See the link in the latest comment above: https://github.com/moggieuk/Happy-Hare/issues/310#issuecomment-4057477518. #310 is still open for Prusa/MMU3 discussion, so I'm closing this question in favour of that one.

If you try that route on v4 and hit something Happy Hare is missing, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk