moggieuk/Happy-Hare #54: Check if proper gate is selected

FOLeader, 2023-09-22:
I would like to have a possibility to check, if the proper gate is selected during the print.
Sometimes it happens, that the encoder is blocked by a very little piece of filament and the wrong color is loaded.
To prevent it I could home the ercf during tool change, but I hope there could be a better solution

thisiscam, 2023-09-22:
I wonder `DUMP_TMC STEPPER=<name>` will show anything interesting if the motor stalled during movement? Maybe that will allow us to detect movement failures.

moggieuk, 2023-09-28:
Unfortunately there is no way to detect "lost steps" so detection would not be possible.  Of source, the "selector touch" operation is to protect against this -- all moves are homing moves and stallguard is used to detect blockage.  If found unblocking operation is attempted.

The periodic re-homing would be a bit of a hack, but you could certain do this by wrapping the `MMU_CHANGE_TOOL` macro.

closed 2024-02-05 by moggieuk