moggieuk/Happy-Hare #55: [mmu_server] Preprocessor throwing errors

PatrikTheDev, 2023-09-23:
I just updated to the latest HH version and enabled the preprocessor. Moonraker seems to have issues starting it though, this is in the logs:

```
2023-09-23 11:10:00,578 [server.py:load_component()] - Unable to load component: (mmu_server)
Traceback (most recent call last):
  File "/home/patrikthedev/moonraker/moonraker/server.py", line 269, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'moonraker.components.mmu_server'
```

Did the moonraker updater not install some files/dependencies?

moggieuk, 2023-09-25:
Did you run `./install.sh` after upgrading as per the announcement?  Sounds like the symlinks to the moonraker module are missing and those are set up with the install.sh script.

PatrikTheDev, 2023-09-26:
Shouldn't Moonraker run that on its own?
[image]

moggieuk, 2023-09-27:
No it doesn't for security reasons .. it's very confusing (confused me) but it only reads that script for dependency checking.
I wish it would run it!    Unless I'm missing something on moonraker config -- if you can make it work let me know!

closed 2024-02-05 by moggieuk