moggieuk/Happy-Hare #57: Feature request: Allow using extruder gears as filament sensor

dflemstr, 2023-09-25:
I reported a similar feature request here: https://github.com/moggieuk/ERCF-Software-V3/issues/97

tl;dr: I really like the pattern of having the extruder gears be the filament sensor, by measuring when the filament gears are pushed apart by the filament. However, this means that the MMU software gets confused by the sensor triggering at exactly the same time as the filament reaches the extruder gears.

After submitting that issue, I switched to the Happy Hare software in this repo instead, in hopes that it would handle this case better. However, I still have issues with the software complaining that filament is still stuck in the sensor after an extruder unload. I wonder if it would be possible to add a specific parameter, to:
1. allow for the filament sensor to be triggered at the same time/sometimes slightly before the filament has reached the extruder gears
2. allow for the filament sensor to still report filament being present at the same time/slightly after the filament has left the extruder gears.

moggieuk, 2023-09-25:
I'm a little confused with the ask.  Happy Hare can use a toolhead sensor to advantage if fitted (this is definitely my preference). On unload the filament will be "reverse homed" to the toolhead sensor and then move the necessary distance to reach the extruder exit (the distance moved include extra).   Without toolhead sensor the extruder will backtrack in large steps (15mm) looking at the encoder to see when the filament is clear.

If you are getting "filament stuck in extruder" it is usually down to too fast unload speeds ( the missed steps causing mismatch in encoder reading) or a malfunctioning encoder.

Additional sensors would be very helpful - a sensor at the entry to the extruder gear probably the most useful -- Happy Hare can support this now but it isn't easy to setup.  It will be soon but of course you need the sensor.

dflemstr, 2023-09-25:
Makes sense that you're confused, I read back my explanation above and realized it didn't make too much sense unless you know my particular set-up.

Hopefully this crude drawing will help:

[image]

I'm talking about having the extruder gears also be fitted with a filament sensor, ie. when the gears are pushed apart, this also triggers a microswitch or similar that detects that filament is in between the gears.  The particular extruder I'm using is this: https://www.printables.com/model/487742-large-gear-clockwork2-ercf-filament-sensor-ecas-bo

I know that this pattern has also been used by Prusa MMU for example, where the filament sensor is triggered by filament between the gears of the extruder.

This is different from detecting filament by crashing into the extruder gears or tugging at the filament and seeing if it moves.  Having the sensor can differentiate between the filament being stuck between the extruder gears (when the filament sensor detects some filament) vs not having any filament at all/the bowden tube being completely empty.

So what might happen during a loading sequence for me, is that the ERCF gear stepper feeds filament to the extruder, and shoves the filament between the filament gears before Happy Hare thinks that the filament should have reached the extruder gears.  It hence gets confused since the filament sensor records filament before the extruder gears are reached (while in fact the extruder gears == the filament sensor).

Similarly, during an unload sequence, Happy Hare might think that the filament should have left the extruder, while in fact the tip of the filament is still slightly pushing the extruder gears apart (even though the extruder isn't "gripping" the filament).  It hence gets confused when reading that filament is still present despite the filament having "left" the extruder.

I have tried manually tweaking the homing distance between the ERCF and the extruder gears so that it thinks the gears are a bit closer to the ERCF than they actually are, and this has mostly fixed the problem for the *loading* sequence, but there are still some false negative sensor readings during the *unloading* sequence.

moggieuk, 2023-11-15:
Sorry long delay, but the simple answer is Yes, Happy Hare v2.2 can now support this.  You would set up as a "extruder" sensor. Here is the conceptual support Happy Hare has for what I call "Type-A" MMUs:

[image]

dflemstr, 2023-11-16:
Amazing, I will try this out. I have since switched to a more "standard" extruder design with a standalone filament sensor after the extruder gears + I plan to add one before the extruder gears, so it might take a while for me to get back to this design. I'm still struggling with #74, and I'm thinking that it will probably be easier to get things to work with a filament sensor rather than extruder sensor for that issue.