moggieuk/Happy-Hare #58: MMU_HOME from BYPASS as last gate position ends with error

AMFkuna, 2023-10-01:
Hi
I have BYPASS block placed after last gate number 8. In mmu_parameters I have total gates set to 8.
If I have selector on BYPASS block and run MMU_HOME, selector goes to GATE 0 but not to endstop.
If I move selector between GATE 8 and BYPASS, then run MMU_HOME, selector moves to endstop and home process run successfully.
example:
GATE 0 selected, MMU_HOME, MMU_SELECT_BYPASS, MMU_HOME, error
GATE 0 selected, MMU_HOME, MMU_SELECT GATE=8, MMU_HOME, success

moggieuk, 2023-10-30:
Need more details.   ERCF v1.1 only has bypass defined in the bearing blocks and that is supported.  Do you have a modified ERCF?   Need more details include how you described your mmu version with vendor and version strings in `mmu_parameters.cfg`.

AMFkuna, 2023-10-30:
Hi
I have this configuration
mmu_vendor: ERCF			# MMU family
mmu_version: 1.1sbt			# MMU hardware version number (add mod suffix documented above)
mmu_num_gates: 8 			# Number of selector gates

If I set mmu_num_gates: to 7 and manualy move the selector to gate 8, MMU_HOME and the selector does not reach the endstop. 
Probably the MMU_HOME macro counts the number of gates from mmu_num_gates, and with MMU_HOME only the length mmu_num_gates * width of the block goes. This probably causes, when the bypass block is at the end, behind the last gate, that it does not reach the endstop.

moggieuk, 2023-11-19:
Ah ok.  Yes, unfortunately I've done my best to keep up with config differences especially for popular mods but I didn't support the bypass block at the end of ERCF v1.1.  Basically you are close to a ERCF v2 should I'd suggest you change the version to that and then I should add the extra room for bypass at the end..

closed 2024-02-05 by moggieuk