moggieuk/Happy-Hare #60: And this is where the Magic doesn't happen...

Durahl, 2023-10-04:
Second attempt in getting the new HHv2 working since the old one now no longer works with the KlipperScreen HappyHare Edition.

Still couldn't get it working most likely because my System is setup differently and not accounted for in either the Automated Installer [nor the Docs](https://github.com/moggieuk/Happy-Hare/blob/main/doc/hardware_config.md#b-hardware-configuration-mmu_hardwarecfg).

**Some Basics about my System:**
- Voron 350 V2.4
- BTT Octopus V1.1
- EBB36 CAN Tool Head
- 6x ERCF ( Sturdy Bunny & Springy using regular Top Hats )

Using the EBB36 CAN Tool Head allows me to free up a second Stepper Slot thus enabling the ERCF to be run entirely off the BTT Octopus V1.1 without the use of another Daughterboard ( EASY-BRD, etc... ). This worked fine on HHv1 including KSHHEd until an Update broke the connection between HHv1 and KSHHEd.

I'm now essentially running HHv1 without a working KSHHEd like a Caveman again.

I've attached a ZIP with my [printer.cfg, EBB_CAN.cfg, and ercf_hardware.cfg](https://github.com/moggieuk/Happy-Hare/files/12810227/Files.zip) since I assume somewhere in one of those three lies the issue - Most likely the EBB_CAN.cfg that has different settings under the `[tmc2209 extruder]` section than the mmu_hardware.cfg asks for to be _transplanted_ into.

I've done a bunch of installs today and sometimes the mmu_hardware.cfg will have a blank
```
[manual_extruder_stepper extruder]
#
```
which if left like that would have Fluidd / Klipper complain that I'm missing stuff like full_steps_per_rotation suggesting that it too _needs_ to be populated.

Other times the mmu_hardware.cfg would ask for the entire `[manual_extruder_stepper extruder]` config right from the getgo with present but undefined Pins and stuff which is kinda in conflict with the claim of the config only requiring the `[tmc2209 extruder]` settings of the Extruder Section to be transplanted.

( populating the Gear and Selector Section doesn't seem to be an issue )

So, yea... Based on the above supplied configs - How should the new config look like while taking the EBB_CAN.cfg and the settings defined in it into consideration?

Also, the EBB_CAN.cfg holds the following Setting:
```
[mcu EBBCan]
canbus_uuid: e8da07817ba6
```
But with _only_ the `[tmc2209 extruder]` portion from it transplanted, can the mmu_hardware.cfg even make use of the stepper?

mdvorak, 2023-10-11:
Follow the manual and instructions, especially here:
https://github.com/moggieuk/Happy-Hare/blob/main/doc/hardware_config.md#b-hardware-configuration-mmu_hardwarecfg
I was also a bit lost at start, but the picture and explanation there are pretty clear.

In the end, my mmu_hardware.cfg section contains this:
```ini
[tmc2209 manual_extruder_stepper extruder]
uart_pin: toolhead:E_TMCUART
interpolate: true
run_current: 0.5
sense_resistor: 0.110
stealthchop_threshold: 0
```

with original tmc config commented out
```ini
# Moved to mmu_hardware.cfg
;[tmc2209 extruder]
;uart_pin: toolhead:E_TMCUART
;interpolate: false
;run_current: 0.5
;sense_resistor: 0.110
;stealthchop_threshold: 0
```

moggieuk, 2023-10-30:
Just got to this.  The quick answer is yes is does work and where you are surprised about klipper and missing parameters is all handled by Happy Hare during the install.

However.  I'm launching an update today that doesn't require the extruder stepper section to be moved.  If you are still struggling then you might want to try again with this release.

Durahl, 2023-11-04:
Uhh... So just head over to https://github.com/moggieuk/Happy-Hare, perform the Install Script and everything ( **ERCF** and **KlipperScreenHHED** ) should work like they did before?

moggieuk, 2023-11-15:
Yes, update the repo (usually via Mainsail).  Because this is a significant release v2.1 -> v2.2 you need to rerun ./install.sh. (no arguments).

Then you should be good.  There is a minor possible problem with the order of the mmu includes and the extruder definition in printer.cfg.  If you see a "microsteps" klipper error on startup then there is info in the discourd #erf_questions channel in the pins. Basically to put the HH includes after the extruder for now.  Will fix that over the weekend.

moggieuk, 2024-02-05:
All the extruder setup is water under the bridge and no longer required so closing this issue.

closed 2024-02-05 by moggieuk