moggieuk/Happy-Hare #62: sync. Extruder and Gear Motor always on

Ronnywa09125, 2023-10-06:
Hello,
I work with the ERFC 1 with synchronized extruder and gear motor. When I extrude manually from the screen, only the extruder operates. Can I change it so that both motors always work in synchronization?

thisiscam, 2023-10-14:
A temporary solution is to pass in `Tx FORCE_IN_PRINT=1 ` when doing toolchangws. You can also overload MMU_TOOLCHANGE to always pass this flag. Until @moggieuk figurws out a revamped solution, this seems to work fine for me

Ronnywa09125, 2023-10-14:
> A temporary solution is to pass in `Tx FORCE_IN_PRINT=1 ` when doing toolchangws. You can also overload MMU_TOOLCHANGE to always pass this flag. Until @moggieuk figurws out a revamped solution, this seems to work fine for me

Thank you

thisiscam, 2023-10-22:
Warning: I discovered there is a problem if we always `force_in_print`. The toolhead might not recover to the correct position if during a MMU_pause another tool changed failed.

closed 2024-02-05 by moggieuk