moggieuk/Happy-Hare #63: Some timenot do Forming tip Macro

Tutankamon00, 2023-10-07:
Hi, some time appear to not start the forming tip macro and eject filament direct and this cause very long hair tip, this appear also for t0-t1-t2 etc command, and also with mmu_eject command, not always but is very frustrating 
[image]

veloyage, 2023-10-10:
I think am having the same issue, it will seemingly randomly not do any tipforming, other times it works fine. No obvious pattern to it regarding gate or anything else I can think of.
I have set force_form_tip_standalone: 1 and am using a filament cutter with form_tip_macro: _MMU_CUT_TIP.

veloyage, 2023-10-10:
From limited testing, this seems fixed in the latest version :)
It complains about nonexistent loading issues now, but thats another story.

Tutankamon00, 2023-10-11:
issue is still here, loaded filament with T0, ok, restart klipper, mmu_eject and NO TIP FORMING!!!! Filament ejected direct!!!! please solve this issue, thank you
[image]

moggieuk, 2023-10-30:
I may some subtle changes which may help you but, you don't have a toolhead sensor... so if the MMU gets into trouble and has to recover position if doesn't have the extra information of whether the filament is in the extruder or not.  Under some conditions it can try other things like moving the extruder (retraction) and looking for encoder measurement, but other times this would be dangerous.   The net result is that sometimes (without toolhead sensor) Happy Hare will conclude something which isn't accurate.

Toolhead sensor is highly recommended if you can fit a reliable one into your extruder.

closed 2024-02-05 by moggieuk