moggieuk/Happy-Hare #65: _MMU_FORM_TIP_STANDALONE variable_parking_distance seems not to work

mdvorak, 2023-10-11:
While it does not cause any problems during normal tool changes, I wanted to use the macro during print init before wipe, to prevent any ooze for best Z calibration.
As I understood the docs and the macro code, filament should be parked to this position regardless of movements performed during tip forming.
This should be accomplished by calling this at macro start

    G91
    G92 E0

and this at macro end

    {% elif PARKING_DISTANCE|int > 0 %}
        G90
        G1 E-{PARKING_DISTANCE} F3000

But my klipper, for some reason, refuses to update extruder position, which in the end causes unload from current position, instead of moving it to given position relative from macro start (adding to the overall retracted distance).
My measured final_pos is 42.5 mm. When i set PARKING_DISTANCE to 42.5, total final distance from the nozzle is 85. When set to 1, final distance is 43.5, etc.. Tested both in print and out of print.

I have honestly no idea, why it behaves this way, because absolute/relative positions change seems fine to me. But it just might be some weird config in my system, no idea.

I have pretty standard Voron 2.4, recently build with Formbot kit, ERCF v1.1 with TrinagleLab kit, and Mellow Fly-SB2040-V2 board on toolhead via CAN. None should influence this.

My full config is here https://github.com/mdvorak/klipper-config/tree/voron2, but its still work-in-progress.

moggieuk, 2023-10-30:
There is indeed a bug in the code that I fixed in my development version but forgot to back port.  Essentially you need to add a `M82`

```
{% elif PARKING_DISTANCE|int > 0 %}
    G90
    M82
    G1 E-{PARKING_DISTANCE} F3000
```

I'm going to update the current version now to fix this.

mdvorak, 2023-10-30:
Hm, I thought G90 would set extruder to absolute mode as well, thanks.

moggieuk, 2024-02-05:
Fixed

closed 2024-02-05 by moggieuk