moggieuk/Happy-Hare #67: Retraction doesn't use manual_extruder_stepper accel value

parkervcp, 2023-10-13:
I have configured my happy hare and can feed to my extruder/hotend just fine, but I cannot get the retraction to work without the stepper jamming. I am using a 6 head kit from trianglelab.

This appears to be the retraction not using the acceleration value set in the mmu_hardware.cfg. I had this issue when feeding to the extruder before I changed the accel.

moggieuk, 2023-10-30:
Which stepper is stalling? (I ask because with all the sync options there are a lot of possibilities).  You can try reducing unload speed.   If you are synced to extruder and doing a retraction then the accel will be controlled by that on the extruder stepper and it is possible the gear stepper cannot keep up.

That said, I have a new release that will be pushed this week that moves away from manual steppers to a second "mmu_toolhead".  It introduces even more control of acceleration.   I'll leave this issue open as "in next release" but I'd appreciate it if you can give me feedback after I release that.

parkervcp, 2023-10-31:
When the `manual_extruder_stepper` runs over around 200 on the accel it stalls. I have it tuned for when it pushes the filament fron the ercf to the hotend. When it pull from the hotend back though it uses a higher acceleration. This is the issue I am having. It should use the same in both directions. I am not syncing the extruder with my ercf.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 the unload acceleration is its own setting (gear_unload_accel in mmu_parameters.cfg), so you can tune it independently of the load acceleration.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk