moggieuk/Happy-Hare #70: Issue with changing filament mid print.

Saikedo, 2023-10-20:
I sometimes do prints where print is not multi material but I need to pause the print, change the filament and resume again. I have been having a lot of issues with Happy-Hare when it comes to that.

Issue #1: When trying to do manual retraction, Happy-Hare does not do a sync with the extruder (I always use sync_to_extruder: 1).

Basically if I pause the print, do MMU_EJECT => MMU_SELECT GATE=n => MMU_LOAD and then try to do a manual extrusion to get rid of the previous color, servo arm is still in down position but MMU is not syncing with the extruder and filament is unable to move.

Issue #2: I tried combating the above mentioned issue by doing MMU_SERVO POS=UP before I do a manual extrusion and then do MMU_SERVO POS=DOWN right before I start printing. There is a very weird issue that happens in this case. First of all, MMU never tries to sync to extruder. Second, it goes to the part to continue printing and then slowly moves away from the printer, raises the hotend up and starts printing like 50mm above the part.

Any idea what might be causing the issue #2 ? Also any plans to address the issue #1 ?

Thank you very much.

thisiscam, 2023-10-22:
For 1, issue is discussed in #62 . Still waiting for @moggieuk to take a look when he gets time.

For 2, you should use `MMU_SYNC_GEAR_MOTOR` for syncing and `MMU_SYNC_GEAR_MOTOR  SYNC=0` for unsyncing.

moggieuk, 2023-10-30:
I'm thinking to just leave Happy Hare in its current state when pausing it will thus be synced when you start manipulating the MMU.  Note that a load operations (like MMU_LOAD, CHANGE_TOOL or Txx) will turn off sync to extruder while it runs through its sequence and will not restore the syncing until you start printing again (unless "force_in_print=1" is passed as a parameter). 

To today you could do:
`MMU_SYNC_GEAR_MOTOR SYNC=1` (turn syncing back on)
`MMU_LOAD` (could result in it being disabled again)
`MMU_SYNC_GEAR_MOTOR SYNC=1` (force syncing back on)
<direct extruder manipulation>
`RESUME`

closed 2024-02-05 by moggieuk