moggieuk/Happy-Hare #71: z-hop not restored after toolchange

thisiscam, 2023-10-20:
Hi,

I noticed that if during a print the MMU got an error and goes into PAUSE, if I called the toolchange `Tx` command during the pause, the toolhead does not restore the lifted z height prior to the change.

thisiscam, 2023-10-22:
I think one case that I made this happen was when I called `MMU_RESET` during a MMU pause. Is it possible to reset only the gates state (buffered or not) and mess with the toolhead position?

moggieuk, 2023-10-27:
MMU_RESET will reset z_hop position!  It is designed to clear all state. 

Try as I have I cannot reproduce this z_hop mismatch.  Every test I run correctly restores to the desired height.  Do you change z_height in any of your macros like PAUSE or FORM_TIP, PRE_UNLOAD, etc...?

thisiscam, 2023-10-27:
I figured out this problem. There were basically two causes:
1. I overwrote TOOLCHANGE to always use `FORCE_IN_PRINT=1`. This caused issues when the printer is paused during a print. I have fixed this issue in my macro now. 
2. For a few times I called `MMU_RESET` causing the z hop position to be overwritten. I think here HH could use some improvements? At least, there should be a way to reset the gate state (e.g. buffered or not) without resetting the saved toolhead position?

moggieuk, 2023-10-30:
There are already existing commands to alter the gate state.  The best is `MMU_GATE_MAP GATE=x AVAILABLE=xx`

Going to close out this issue.

closed 2023-10-30 by moggieuk