moggieuk/Happy-Hare #73: Filament winding with stepper

Turboralle83, 2023-10-21:
Hello, is it possible to control a stepper to roll up filament during mmu_unload_bowden-macro? If not that would be my only suggestion for improvement.
Otherwise a great program

thisiscam, 2023-10-22:
You can make the stepper a `manual_stepper`, and sync it to the toolhead's motion queue during the unloading.

Turboralle83, 2023-10-22:
Thanks for the quick answer!
But the extruder moves only until the filament is parked over the wheels. During the fast bowden unload macro the extruder stands still.
I want the roller to rotate only during the fast move.
[image]

thisiscam, 2023-10-22:
I see.. I think this is not possible right now.

I think it makes sense for HHv2 to keep the invariant that whenever filament is moved, by either the extruder or the gear stepper, the extruder motion queue is used and the actual steppers are only synched/unsynced accordingly. 
This way, your use case can be handled by syncing the winding stepper to the extruder's motion queue.

Turboralle83, 2023-10-22:
This implementation would be great!
I think then you could build a system like the AMS from it and save the space for the filament buffers.
I think the community would celebrate that!

thisiscam, 2023-10-22:
I think the current toolhead branch is making progress towards that direction. It's probably easier to implement this feature once that gets rolled out

moggieuk, 2026-10-21:
Thanks for the idea, and sorry for the very slow reply. Happy Hare v4 (v4.1 is the current release) has this as the "eSpooler", so I'm closing this.

An eSpooler is a DC motor (usually a small N20, PWM controlled) on each spool. Happy Hare drives it with the gear motor, so it rewinds the spool during unloads, including the fast bowden move, and can also assist on loads. Its speed scales with the gear stepper speed.

To set it up:
- In menuconfig, go to "MMU Features / Additions", enable "Has eSpooler?" and set the motor pins in the eSpooler config.
- Tune it with the `espooler_*` settings in `mmu_parameters.cfg`. `espooler_operations` picks which modes are active (`rewind`, `assist`, `print`).
- `MMU_ESPOOLER` lets you test and control it by hand.

One difference from what you described: it's designed for DC motors, not a Klipper stepper. If you specifically need a stepper-driven winder, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk