moggieuk/Happy-Hare #74: Possibly incorrect calculation of unload filament parking position

dflemstr, 2023-10-25:
With the latest Happy Hare code (v2.1.0-26-gae84c581) I'm getting this error 80% of the times that I try unloading filament:

```
After tip formation, extruder moved (park_pos): 40.00, encoder measured 72.00
MMU [T0] <<< [En] <<<<<<< [Ex] << [Ts] <. [Nz] (@40.0 mm)
An issue with the MMU has been detected whilst out of a print
Reason: Unload sequence failed: Filament seems to be stuck in the extruder.
Occured when changing tool: T0 > T1
```

My encoder is very well calibrated so it is surprising that the difference between the actually moved value and the encoder measured value is so large.

I suspect that the issue is that the encoder value is snapshotted here:

https://github.com/moggieuk/Happy-Hare/blob/ea39e0604878aacf0c49fe97cd9f970516f35783/extras/mmu.py#L3833

...but because the `_MMU_FORM_TIP_STANDALONE` moves the filament back and forth so much, the encoder value will likely be widely inaccurate because of the constant change of direction leading to sample aliasing.

This error gets propagated here, causing the final unload move to be too short and for the error to eventually be reported:
 
https://github.com/moggieuk/Happy-Hare/blob/ea39e0604878aacf0c49fe97cd9f970516f35783/extras/mmu.py#L3507C57-L3507C57

I suppose that this won't happen if the `output_park_pos` variable is set, which might be the reason why everyone isn't seeing this error?

dflemstr, 2023-10-25:
I should add that setting:

```
variable_output_park_pos: 20           # The park position (output variable passed back to Happy Hare)
```

...in the `[gcode_macro _MMU_FORM_TIP_STANDALONE]` seems to fix the issue for me. However it seems like this isn't intended as a viable workaround because the variable is no longer there in the latest version of the `mmu_software.cfg` file in this repo?

moggieuk, 2023-10-27:
The encoder reading is likely correct - remember the encoder only counts in one direction but a tip forming move moves backwards and forwards.  The info is presented in the debug message but NOT used to determine park position.  The park position is based on the aggregate actual stepper movement.

The variable_output_park_pos should be `-1` or not set at all which tells Happy Hare to believe the stepper movement.  If you set it that it will be used instead of stepper movement.

What you need to do is look in the `mmu.log` (or read the DEBUG level of logging) to see the cause of the error.  It could be the dimensions of the extruder aren't quite right or most likely, your movement speeds are too fast and you are loosing steps.

moggieuk, 2023-10-27:
Without the `mmu.log` I can't help further, but I'm very confident this is not a bug in Happy Hare logic.

dflemstr, 2023-10-27:
Because I haven't been able to narrow down what's going on here, I'll close this issue for now until I've determined the root cause. Apologies for the noise!

dflemstr, 2023-11-17:
I had some time to get back to this issue again now, because I don't have a backlog of stuff I need to print and where I don't mind some manual intervention to get the MMU kicked off for those prints :)

I've also made sure to update to the latest Happy-Hare version and switched to config that is maximally similar to what you get from the `./install.sh` default templates.

Here's the latest log from when an unload failed:
[mmu.log](https://github.com/moggieuk/Happy-Hare/files/13385616/mmu.log)
...and the 
[mmu_parameters.cfg](https://github.com/moggieuk/Happy-Hare/files/13385626/mmu_parameters.txt) config (renamed to `.txt` because of GitHub upload limitations)

To be clear I'm running a TradRack + ERCF encoder hybrid, and hence there's both an encoder, gate filament sensor, and toolhead filament sensor. I plan to add an extruder filament sensor too if that helps narrow down issues like this.

I tried to enable extruder syncing whenever possible to reduce the encoder error due to bowden tube wiggleroom, but let me know if/when this is ill-advised.

The key issue seems to be with:

```
01:46:58 - DEBUG: Reverse homing up to 25.0mm to toolhead sensor (synced)
...
01:47:01 - - TRACE: Reverse homing to toolhead sensor. Stepper: 'gear+extruder' did not home after moving -25.0mm (of max -25.0mm), encoder measured 26.0mm (delta -1.0mm). Pos: @-65.0, (66.0mm)
```

It seems like the 25mm gets calculated here:

https://github.com/moggieuk/Happy-Hare/blob/3d41ecc7d12f0e421f489b7af53166f71aa4c27b/extras/mmu.py#L3135C13-L3135C105

This made me realize that tweaking `toolhead_unload_safety_margin` might be a decent way of solving this problem too, but it is strange if this would be needed for my case, since I use a very standard toolhead design (clockwork 2/stealthburner).

Digging into the other options of how to affect the `length` calculation, it seems the options are:

  * Tweaking `toolhead_sensor_to_nozzle`: I don't want to increase this too much since it leads to nozzle oozing on tool changes.
  * Tweaking the `park_pos`, which as far as I can tell is computed from the `_form_tip_standalone` sequence.

Going down the `park_pos` route, since I have not set any parking position manually, it seems the code will end up taking this path:

https://github.com/moggieuk/Happy-Hare/blob/3d41ecc7d12f0e421f489b7af53166f71aa4c27b/extras/mmu.py#L3431

Now the question is, is there any way that this calculation could be drastically off?  It seems very weird to me that this value might be more than 25mm off, leading to the toolhead filament sensor homing sequence to fail?

thisiscam, 2023-11-17:
> Now the question is, is there any way that this calculation could be drastically off? It seems very weird to me that this value might be more than 10mm off, leading to the toolhead filament sensor homing sequence to fail?

 You are using tip forming instead of cutter, right? If that's the case, it might be that tip forming is losing steps.

dflemstr, 2023-11-17:
> it might be that tip forming is losing steps

Yes maybe, however I don't see any indication of that, ie no noise, no grinding sound, etc...

thisiscam, 2023-11-19:
> Yes maybe, however I don't see any indication of that, ie no noise, no grinding sound, etc...

It might be losing steps when the extruder rapidly changes direction. That is quite hard to hear. What if you try to reduce the speed and number of moves of tip forming? Is the measured distance still off? 

AFAIK, if you just used standard gcode movements in your tip forming, that measured distance should be incorrect. So losing steps would be my top guess.

moggieuk, 2023-11-19:
Happy Hare definitely will measure the actual extruder stepper movement and set park position correct when forming tips.  It is almost certainly a result of rapid movements your extruder can't handle.   Try reduced the speeds by half as a test.  Make sure you extruder is gripping filament and (recommend) use the % current increase to temporarily increase current to extruder when doing tip forming -- I run at 120% for this move.

closed 2024-12-26 by github-actions[bot]