moggieuk/Happy-Hare #76: Properly support Toolhead sensor ABOVE extruder (SImilar to Prusa MMU)

Alia5, 2023-10-29:
Much like a Prusa machine with MMU, my Voron currently has a toolhead sensor above the extruder_gears.
Currently, the only option is to not use the toolhead sensor with HappyHare at all.

Instead, I imagine it working much like the Prusa MMU as well.

**Load**

1. Load filament fast to extruder sensor.
2. Slow load an additional, specified length (sensor to extruder distance) in sync
3. Continue with the already existing loading operation after gears.

**Unload**

1. Slow unload in sync until the filament passes the sensor.
2. Fast Unload
...

---

This would minder the reliance on a good working encoder as well as be compatible with more toolheads and toolhead sensors out there.  
Additionally, the original Prusa MMU2S expects a toolhead sensor above the extruder gears, much like this one.  
This would ease the workload on supporting the original Prusa MMU.

moggieuk, 2026-10-21:
Thanks, and sorry for not closing this out earlier. This is supported in Happy Hare v4 (v4.1 is the current release), so I'm closing it.

v4 has a dedicated "extruder" (entry) sensor for exactly this: a sensor just before the extruder gears. It can be used alongside a normal toolhead sensor after the gears, or instead of one.

To set it up:
- In menuconfig, open "Toolhead sensors/settings", enable "Has extruder (entry) sensor?" and give the pin. It ends up as `extruder_switch_pin` in the `[mmu_toolhead ...]` section of `mmu.cfg`.
- Set `toolhead_entry_to_extruder` to the distance from the sensor to the extruder gears.
- Set `extruder_homing_endstop: extruder`. The fast bowden load then stops short of the sensor and homes to it before the toolhead load continues.

If there's still something about this setup that v4 doesn't handle, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk

Alia5, 2026-10-21:
Ouh wow! That came unexpected

Thanks! :D