moggieuk/Happy-Hare #77: Toolhead Board (Ebb36) Diag pin Sensorless  "touch" homing?

mochoandre, 2023-10-31:
Is it possible to use extruder  diag pin in EBB36 so i can use the "touch" fuction in homing to nozzle in new HAPPY-HARE V2?

mrmultifunk, 2023-11-04:
I don't think the ebb36 has the diag pin connected. I can look it up in the schematic when i'm back on the pc.

mochoandre, 2023-11-05:
I Also think that but just checking if someone and a better in depth information or a workaround

Surion79, 2023-11-06:
The diagrams on EBB36 and EBB42 do not show a DIAG pin defined, only ENABLE, DIR, STEP, UART. 
EBB SB2240 and SB2209 have DIAG pins.

mochoandre, 2023-11-07:
Thank you. I have a EBB SB2240 but can not use it with my lgx lite

moggieuk, 2023-11-14:
This doesn't sound like an issue. Can I close this out?

BTW the "home to nozzle" functionality is pretty cool but definitely experimental and needs careful tuning of stall guard to get the just right.

closed 2024-02-05 by moggieuk