moggieuk/Happy-Hare #79: _MMU_POST_UNLOAD

FOLeader, 2023-11-04:
We have _MMU_POST_LOAD and _MMU_PRE_UNLOAD in software.cfg.
For I am useing the filament cutter form kevinakasam which is directly under the encoder, I would like to cut the filament directly after unload. I tried to add a section _MMU_POST_UNLOAD, but I had to add a ercf_eject into the section. means it is not working after unloading, but as unload and cut. could you please change it that _MMU_POST_UNLOAD is really after filament is unloaded before tool change

Xonman, 2023-11-05:
@FOLeader this isn't a proper fix, but I got around this by replacing the eject with a mmu_recover in the cutter macro.

FOLeader, 2023-11-10:
Thx, I will try, when I got a replacement for my SB2040 :(

moggieuk, 2023-11-14:
The correct fix is really not to try a RECOVER/EJECT, but rather:
```
_MMU_STEP_SET_FILAMENT STATE=1
MMU_EJECT
```

Check out the doc here: https://github.com/moggieuk/Happy-Hare/blob/main/doc/gcode_customization.md#---_mmu_load_sequence--_mmu_unload_sequence

closed 2024-03-11 by moggieuk