moggieuk/Happy-Hare #80: Configuration without EASY-BRD etc

mrdamianopl, 2023-11-05:
Hello,
I wanted to change from ERCF to Happy Hare but I have a problem.

I have an octopus board with free stepper motor sockets for ERCF and this is how it is connected, and it was working with ERCF

During the installation process, the HH wizard asks me whether I am using EASY-BRD etc. I select no, 
Then asks me about the serial port for the MMU MCU and this is where I encounter a problem
First one is just copy of my octopus serial suggested by the wizard and accepted, the second one is not accepted

Klippy log below

`mcu 'mmu': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/serial/by-id/usb-Klipper_stm32f446xx_22000E000650534E4E313020-if00: [Errno 11] Resource temporarily unavailable`

`mcu 'mmu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM1: [Errno 2] No such file or directory: '/dev/ttyACM1'`

What did I do wrong or what did I forget? I've been struggling with this for almost two days now

closed 2023-11-12 by mrdamianopl

Wolffy00, 2023-12-24:
How do you resolve you problem
Same by me
Ercf work fine
HH V2 unknown how to set up for direct wiring on btt m8p

mrdamianopl, 2023-12-25:
comment out all mmu.cfg (mcu mmu and board_pins mmu)
check pinout in mmu_hardware

mrdamianopl, 2023-12-25:
Comment out all mmu.cfg (mcu mmu and board_pins mmu)
check pinout in mmu_hardware

Durahl, 2024-04-26:
@mrdamianopl Can you provide a code example of all the things that have to be comment out?

I'm having the same problem with running my ERCF through my BTT Octopus V1.1 after changing from legacy HH to the new one but no matter what I comment out [there's always an Error Message being displayed](https://github.com/moggieuk/Happy-Hare/issues/282#issuecomment-2077965127).

Wolffy00, 2024-04-26:
here is my functional configuration with Manta M8P + ERCF directly wired to the MB (no easy bird or else ...)
Ratos installed (Ratrig Klipper version)
[mmu.cfg.txt](https://github.com/moggieuk/Happy-Hare/files/15135490/mmu.cfg.txt)
[mmu_hardware.cfg.txt](https://github.com/moggieuk/Happy-Hare/files/15135491/mmu_hardware.cfg.txt)
[mmu_parameters.cfg.txt](https://github.com/moggieuk/Happy-Hare/files/15135492/mmu_parameters.cfg.txt)
[printer.cfg.txt](https://github.com/moggieuk/Happy-Hare/files/15135493/printer.cfg.txt)

Durahl, 2024-04-28:
@Wolffy00
[image]
The files provided _much_ needed insight - I'm finally up and running again.