moggieuk/Happy-Hare #82: Unload callback called after unload failure

Xonman, 2023-11-05:
If a `_MMU_POST_UNLOAD` callback macro is defined it will be run even if unload fails due to filament stuck in the Bowden tube. Given the filament hasn't been unloaded I don't think the macro should run in this scenario.

moggieuk, 2023-11-14:
Thanks I'll look at this. This was added quite hastily to support filament cutter at the MMU but you are right in that it should only be run on successful unload of the bowden.

moggieuk, 2023-11-19:
Hi. You can going to have to help me with this.  I've reviewed the logic and the MMU_POST_UNLOAD macro should not get called if there is an error earlier.  Can you (i) make sure your Happy Hare is up-to-date, (ii)send the `mmu.log` where this occurs so I can review the flow...

Xonman, 2023-12-05:
I've not actually been able to replicate this since it happened the first few times. There's a strong possibility it was actually my macro's fault rather than HH. I'll close, thanks for investigating.

closed 2023-12-05 by Xonman