moggieuk/Happy-Hare #84: Internal error on command: "MMU_CHECK_GATES"

sobsmb, 2023-11-10:
Voron V2.4

happy-hare: v2.1.0-42-g8436a9df  

During pause due to clog. issued a check gate command and received:

-----------------------------------------------------------
Internal error on command:"MMU_CHECK_GATES"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
-----------------------------------------------------------

klipper: v0.11.0-311-g9e765dae

All other software up to date.

Let me know if you want anything else.

moggieuk, 2023-11-14:
Oh no.  Do you still have the klippy.log file?  If you ever get `internal error` then all I need will be in that log file..

moggieuk, 2023-11-19:
I believe this was fixed in v2.2 ..

sobsmb, 2023-11-19:
Sorry for the delay I was stuck in Jury Duty, and so many things took a back seat.  I haven't done a multi-color print since and unfortunately do not have that file anymore.  I will update and try to do the same thing and report back

sobsmb, 2023-11-19:
I tested this and can confirm this is no longer a issue with v2.2   Thanks

closed 2023-11-19 by sobsmb