moggieuk/Happy-Hare #86: Option 'microsteps' in section 'extruder' must be specified

kerkmann, 2023-11-13:
Since the newest version `v2.2.0` I am getting the error message `Option 'microsteps' in section 'extruder' must be specified
`. The strange part is, the printer is working without Happy Hare. But when I install it (doesn't matter if I migrate or do a clean install), I am getting the error message `Option 'microsteps' in section 'extruder' must be specified
`. :(

What information do you need and has someone the same issue?

shaneshort, 2023-11-13:
I'm having the exact same issue after upgrading this morning.

shaneshort, 2023-11-14:
@kerkmann after spending a lot of time digging around in code I don't quite understand, it looks like the immediate fix for me, is to include the mmu configuration *after* the extruder is defined in the config. I originally had it at top.

 @moggieuk from what I can understand a 'extras.mmu_toolhead.MmuExtruderStepper' object is instantiated by HappyHare which inherits from the extruder. I can see there's a callback function registered to insert this object once klippy connects, but It seems if this is instantiated before the extruder is, it errors.

moggieuk, 2023-11-14:
Yes. this logic is tricky and there was a reason for the order of things but clearly it is generating problems for folks.  The workaround (until I get back home to my printer .. in about 3 days) is to move the `[include mmu/base/*.cfg]` to AFTER the extruder definition in the printer.cfg file.

kerkmann, 2023-11-14:
@moggieuk thank you very much for your quick response. Can we add it to the README or the install script somewhere? Otherwise it's a little bit misleading. :)

Thank! :heart:

kerkmann, 2023-11-15:
Can confirm, moving them under the `extruder` definition is solving the problem. It was a good step to force me to split my `printer.cfg`. Now it's much cleaner. :P

```cfg
[include printer/base.cfg]
[include printer/mcu.cfg]

[include printer/bed_mesh.cfg]
[include printer/display.cfg]
[include printer/extruder.cfg]
[include printer/fan.cfg]
[include printer/heater_bed.cfg]
[include printer/homing.cfg]
[include printer/input_shaper.cfg]
[include printer/led.cfg]
[include printer/probe.cfg]
[include printer/stepper_x.cfg]
[include printer/stepper_y.cfg]
[include printer/stepper_z.cfg]
[include printer/temperature_sensor.cfg]

[include mmu/base/*.cfg]
[include mmu/optional/client_macros.cfg]
[include mmu/optional/mmu_menu.cfg]

[include macros.cfg]
[include mainsail.cfg]
[include nevermore.cfg]
[include timelapse.cfg]
```

closed 2024-02-05 by moggieuk