moggieuk/Happy-Hare #87: Internal error on command:"MMU_CHANGE_TOOL"

smilecaozheng, 2023-11-15:
Scenario: When the printer is in a ready state, I used [MMU_SYNC_GEAR_MOTOR] command to extrude some filaments, and then I started printing normally, reporting the error

Internal error on command:"MMU_CHANGE_TOOL"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

klippy.log

Receive: 80 7125.793869 7125.789575 18: seq: 1a, tmcuart_response oid=0 read=b'\n\xfa\xef-\xc0\x14\x02\x08 \xfa'
Receive: 81 7125.800695 7125.796411 18: seq: 1b, tmcuart_response oid=0 read=b'\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 82 7125.826511 7125.825345 8: seq: 1d, neopixel_result oid=5 success=1
Receive: 83 7125.827731 7125.825345 14: seq: 1d, analog_in_state oid=9 next_clock=4165502001 value=6637
Receive: 84 7125.837968 7125.825345 15: seq: 1d, analog_in_state oid=10 next_clock=4165622001 value=29178
Receive: 85 7125.856120 7125.855217 8: seq: 1f, neopixel_result oid=5 success=1
Receive: 86 7125.888195 7125.883193 14: seq: 11, analog_in_state oid=15 next_clock=4166222001 value=1590
Receive: 87 7125.889720 7125.889037 8: seq: 12, neopixel_result oid=5 success=1
Receive: 88 7125.902891 7125.902576 11: seq: 13, clock clock=4162890425
Receive: 89 7125.920988 7125.920265 8: seq: 15, neopixel_result oid=5 success=1
Receive: 90 7125.949667 7125.948978 8: seq: 17, neopixel_result oid=5 success=1
Receive: 91 7125.980064 7125.979366 8: seq: 19, neopixel_result oid=5 success=1
Receive: 92 7126.010744 7126.010020 8: seq: 1b, neopixel_result oid=5 success=1
Receive: 93 7126.021144 7126.010020 14: seq: 1b, trsync_state oid=8 can_trigger=1 trigger_reason=0 clock=4164305951
Receive: 94 7126.030927 7126.021411 14: seq: 1c, trsync_state oid=8 can_trigger=1 trigger_reason=0 clock=4164425892
Receive: 95 7126.041176 7126.031689 14: seq: 1d, trsync_state oid=8 can_trigger=1 trigger_reason=0 clock=4164545863
Receive: 96 7126.050942 7126.041850 14: seq: 1e, trsync_state oid=8 can_trigger=1 trigger_reason=0 clock=4164665858
Receive: 97 7126.061021 7126.051854 14: seq: 1f, trsync_state oid=8 can_trigger=1 trigger_reason=0 clock=4164785858
Receive: 98 7126.070918 7126.061400 14: seq: 10, trsync_state oid=8 can_trigger=1 trigger_reason=0 clock=4164905857
Receive: 99 7126.073174 7126.072258 12: seq: 12, shutdown clock=4164927178 static_string_id=Command request
Internal error on command:"T0"
virtual_sdcard dispatch
Traceback (most recent call last):
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 3960, in _wrap_sync_extruder_to_gear
    yield self
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 3739, in _trace_filament_move
    trig_pos = hmove.homing_move(pos, speed, probe_pos=True, triggered=homing_move > 0, check_triggered=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/klipper/klipper/klippy/extras/homing.py", line 94, in homing_move
    self.toolhead.drip_move(movepos, speed, all_endstop_trigger)
  File "/home/klipper/klipper/klippy/toolhead.py", line 486, in drip_move
    self.move_queue.flush()
  File "/home/klipper/klipper/klippy/toolhead.py", line 176, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/klipper/klipper/klippy/toolhead.py", line 333, in _process_moves
    self._update_drip_move_time(next_move_time)
  File "/home/klipper/klipper/klippy/toolhead.py", line 467, in _update_drip_move_time
    self._update_move_time(npt)
  File "/home/klipper/klipper/klippy/toolhead.py", line 293, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/klipper/klipper/klippy/mcu.py", line 969, in flush_moves
    raise error("Internal error in MCU '%s' stepcompress"
mcu.error: Internal error in MCU 'sb2040' stepcompress

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/klipper/klipper/klippy/extras/virtual_sdcard.py", line 264, in work_handler
    self.gcode.run_script(line)
  File "/home/klipper/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/klipper/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/klipper/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/klipper/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/klipper/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/klipper/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/klipper/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 4329, in cmd_MMU_CHANGE_TOOL
    if self._change_tool(tool, self._is_printing(force_in_print), skip_tip):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 4138, in _change_tool
    self._select_and_load_tool(tool)
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 4066, in _select_and_load_tool
    self._load_sequence()
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 3234, in _load_sequence
    self._load_extruder()
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 3088, in _load_extruder
    _,homed,_,_ = self._trace_filament_move("Homing to toolhead sensor", self.toolhead_homing_max, motor=motor, homing_move=1, endstop_name=self.ENDSTOP_TOOLHEAD)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 3730, in _trace_filament_move
    with self._wrap_sync_extruder_to_gear(motor in ["gear+extruder", "extruder"], extruder_only=(motor == "extruder")):
  File "/usr/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/home/klipper/klipper/klippy/extras/mmu.py", line 3963, in _wrap_sync_extruder_to_gear
    self.mmu_toolhead.sync_extruder_to_gear(None)
  File "/home/klipper/klipper/klippy/extras/mmu_toolhead.py", line 210, in sync_extruder_to_gear
    self.flush_step_generation()
  File "/home/klipper/klipper/klippy/toolhead.py", line 338, in flush_step_generation
    self.move_queue.flush()
  File "/home/klipper/klipper/klippy/toolhead.py", line 176, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/klipper/klipper/klippy/toolhead.py", line 333, in _process_moves
    self._update_drip_move_time(next_move_time)
  File "/home/klipper/klipper/klippy/toolhead.py", line 467, in _update_drip_move_time
    self._update_move_time(npt)
  File "/home/klipper/klipper/klippy/toolhead.py", line 293, in _update_move_time
    m.flush_moves(mcu_flush_time)
  File "/home/klipper/klipper/klippy/mcu.py", line 969, in flush_moves
    raise error("Internal error in MCU '%s' stepcompress"
mcu.error: Internal error in MCU 'sb2040' stepcompress
Exiting SD card print (position 91695)
Dumping stepper 'stepper_mmu_gear' (mmu) 1 queue_step:
queue_step 0: t=4164197337 p=90046 i=7717196 c=1 a=0
Dumping stepper 'extruder' (sb2040) 1 queue_step:
queue_step 0: t=4164340824 p=671545 i=540112896 c=1 a=0
Requested toolhead position at shutdown time 347.050371: (13.0, 32.0, 5.171796403572042)

moggieuk, 2023-11-15:
Ok. I think I know what this is...

Is your microstep size for gear stepper 8?    If so set to 16... 

Let me explain.  When steppers are synced and performing a homing move they both move together in small steps.  These steps are invisible to you but it is what klipper calls "drip" moves.  If a move is so small that one stepper of the synchronized pairs does not have the resolution to move, it will later see this as a "step compress" internal error.  I believe it is an oversight in Klipper but materializes in Happy Hare because we are synchronized two very different steppers - the gear on the MMU and the extruder.  The extruder is usually set up to be high resolution whilst the gear is not.

smilecaozheng, 2023-11-16:
I understand. I used 64 microsteps for the extruder, and 16 microsteps for the gear motor, so

smilecaozheng, 2023-11-16:
#	In E0-MOT Position					            
[tmc2209 extruder]                                      
uart_pin: sb2040:gpio8                                  
run_current: 0.350                                      
stealthchop_threshold: 0  		                        

[extruder]                                              
step_pin: sb2040:gpio9                                  
dir_pin: !sb2040:gpio10                                 
enable_pin: !sb2040:gpio7                               
microsteps: 64								            
full_steps_per_rotation: 200  			                
gear_ratio: 50:10							            
rotation_distance: 22.6556595				             
pressure_advance_smooth_time: 0.001                      

nozzle_diameter: 0.400						            
filament_diameter: 1.75						            

heater_pin: sb2040:gpio6                                
min_temp: -20                                           
max_temp: 300                                           
max_power: 1.0								            
min_extrude_temp: 170					                
sensor_pin: sb2040:gpio27                               

## PT1000
# sensor_type: PT1000                                   
# pullup_resistor: 1000                                 

## NTC3950 100K
sensor_type: NTC 100K MGB18-104F39050L32

smilecaozheng, 2023-11-16:
```
#	In E0-MOT Position					            
[tmc2209 extruder]                                      
uart_pin: sb2040:gpio8                                  
run_current: 0.350                                      
stealthchop_threshold: 0  		                        

[extruder]                                              
step_pin: sb2040:gpio9                                  
dir_pin: !sb2040:gpio10                                 
enable_pin: !sb2040:gpio7                               
microsteps: 64								            
full_steps_per_rotation: 200  			                
gear_ratio: 50:10							            
rotation_distance: 22.6556595				             
pressure_advance_smooth_time: 0.001                      

nozzle_diameter: 0.400						            
filament_diameter: 1.75						            

heater_pin: sb2040:gpio6                                
min_temp: -20                                           
max_temp: 300                                           
max_power: 1.0								            
min_extrude_temp: 170					                
sensor_pin: sb2040:gpio27                               

## PT1000
# sensor_type: PT1000                                   
# pullup_resistor: 1000                                 

## NTC3950 100K
sensor_type: NTC 100K MGB18-104F39050L32          
```

smilecaozheng, 2023-11-16:
```
[tmc2209 stepper_mmu_gear]
uart_pin: mmu:gpio9                                 
interpolate: true                                   
run_current: 0.5			                        
hold_current: 0.1                                   
sense_resistor: 0.110                               
stealthchop_threshold: 0		                    

[stepper_mmu_gear]
step_pin: mmu:gpio7                                 
dir_pin: !mmu:gpio8                                 
enable_pin: !mmu:gpio6                              
gear_ratio: 50:17                                   
microsteps: 16                                      
full_steps_per_rotation: 200                        

rotation_distance: 22.213836                        
```

smilecaozheng, 2023-11-16:
So let me change the extruder to 16 microsteps and test it again

Grafton1046, 2023-11-17:
Having the same error, didn't think about where I changed micro steps on the gear motor, testing that fix as well

moggieuk, 2023-11-19:
I'm interested in a little more information... it shouldn't be necessary to match the microsteps for the extruder and gear stepper, but just to have enough resolution for the gear_stepper.

Can you summerize what didn't work and what did work for me.  It would be very helpful.

Grafton1046, 2023-11-19:
> I'm interested in a little more information... it shouldn't be necessary to match the microsteps for the extruder and gear stepper, but just to have enough resolution for the gear_stepper.
> 
> Can you summerize what didn't work and what did work for me. It would be very helpful.

Sure, short version:
Got annoyed at the Ercf easy brd, due how it was a pain to update kipper. So I swapped the board with a fystec burrow board, it has buttons to get into dfu... it also has tmc 2209's vs the tmc2226's that came with my easy brd.

 Upon swapping the pin aliases and boards, I notoced I couldn't hit the same feeds and speeds with the gear motor as before. 

So I lowered speeds and microsteps to 1 to get more torque. Didn't have issues on single materal prints,  only noticed when I would try multi natural prints it would crash klipper. 

Now I'm waiting on a mellow canbus ercf board because the rp2040 doesn't like to wait on the pi before giving up on usb...

moggieuk, 2023-11-19:
Ah ok, thanks.  Microsteps will need to be 16 to be able to sync with the extruder on homing moves.  BTW I'm not sure you get much more torque with no microstepping and potentially you could be more likely to miss steps.  What current are you running your stepper at and what kind of stepper?  I have the BOM Nema 14 and can run at 250mm/s without issue.   I have minimized friction though.

Grafton1046, 2023-11-19:
> Ah ok, thanks. Microsteps will need to be 16 to be able to sync with the extruder on homing moves. BTW I'm not sure you get much more torque with no microstepping and potentially you could be more [...]

Running a stepper online nema 17x 48 at 1.25amps, I've not figured out my friction issue, sadly. The extruder sync has been a god send, love that feature.

Here is my ercf/buffer setup https://imgur.com/a/FleGvUW on top of a v2.4 350

LostFool, 2023-12-20:
I have gear at 16 but get same error.
[image]
[mmu.log](https://github.com/moggieuk/Happy-
[klippy.log](https://github.com/moggieuk/Happy-Hare/files/13732084/klippy.log)
Hare/files/13732031/mmu.log)
[config-20231214-201930.zip](https://github.com/moggieuk/Happy-Hare/files/13732036/config-20231214-201930.zip)

moggieuk, 2024-02-05:
Try increasing the gear stepper `microsteps` to 32.  The problem is in the mismatch of step size but i depends on the config of your extruder to know what is a good size for you..

LostFool, 2024-02-24:
Updated to 32 microsteps. A lot has changed firmware wise as well since the original issue. I haven't had the issue again yet. Hoping the 32 microsteps did it, other wise I do not know. I know after some firmware updates I started having more issues solved.

moggieuk, 2024-03-11:
I would like to close this issue since the only know issue with stepcompress error is when move difference for one microstep is grossly mismatched between gear and extruder. Steppers. This is a klipper issue but I yet to find a scenario that can’t be fixed by adjusting microsteps of one of the steppers.

LostFool, 2024-03-11:
I am no longer having this issue. Either because I changed my extruder and gear stepper to 32, or by other updates.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

This turned out to be a microstep resolution mismatch between the gear and extruder steppers, and raising the gear stepper microsteps fixed it for everyone who reported it.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk