moggieuk/Happy-Hare #88: Synchronized load does not detect unsuccesful load

BCRans, 2023-11-16:
Hi, 

first of all thank you for this SW, I have could not write all this code myself. 

I noticed since I switched to the new Happy Hare (v2.2 now), that sometimes the machine misbehaves. Sometimes when the servo did not push down with enough force, not enough filament was inserted, did not reach extruder gears. The load macro continued to try to home to extruder (by collision), did not reach gears, still continued the sequence, extruder did not pick up filament, yet the SW evaluated as an OK load. 

To test it I disconnected the bowden from the ERCF, and I have 1m+ filament dangling around the side of the printer, with the software thinking the load was succesful. Could you please check? I think the root cause is, that during the final load both the ERCF and extruder gears are moving the filament together, so all movements are registered by the encoder, evaluating as an OK load.

Idea: is it possible to disable the "home to extruder" step? the encoder is precise enough (0,7mm on my setup) to tell if the filament is close enough to extruder gears. Then combined move of let's say 10mm (option to set length + if user wants to disable the homing would be nice), disengage servo, pull with extruder another 10mm to check if filament is really loaded, then final push into nozzle. There was an option in v2.1 mimicking this behavior called toolhead_sync_load, now not to be found in the parameters.cfg, but homing still stayed. On the even older V3 version I was able to disable this homing step altogether by setting ERCF_TEST_CONFIG home_to_extruder=0 in my start Gcode and it worked reliably. Also the homing step without "sensorless homing" on the gear motor often grinds the filament, to have another reason to remove this step.

Note: The main issue with this system is the too weak servo - I have the MG90 servo now installed. Fighting with kickback. I also have your springy mod, the servo is able to grip the filament only when I take the spring adjustment screw completely out. I ordered the Savox servo, I hope I will get it this year... 

I have found 2 other bugs in the unload process, these are much harder to reproduce, I will try, maybe I am not alone with those ones.

Thank you!

Log with debug option set of the last step: 
```
20:52:55 MMU [T2] >>> [En] >>>>>>> [Ex] ... [Nz] 940.6mm (e:936.1mm)
20:52:55 - DEBUG: Homing to extruder gear, up to 50.0mm in 2.1mm steps
20:52:55 Modifying MMU gear stepper run current to 35% for collision detection
20:52:56 - - TRACE: _movequeues_wait_moves(toolhead=True, mmu_toolhead=True)
20:52:56 - - - STEPPER: GEAR: dist=2.1, speed=50.0, accel=400.0, sync=False, wait=False
20:52:56 - - TRACE: _movequeues_wait_moves(toolhead=True, mmu_toolhead=True)
20:52:56 - - TRACE: Homing step #1. Stepper: 'gear' moved 2.1mm, encoder measured 0.7mm (delta 1.4mm). Pos: @942.7, (936.8mm)
20:52:56 - DEBUG: Extruder entrance found after 2.1mm move (1 steps), encoder measured 0.7mm (delta 1.4mm)
20:52:56 Restoring MMU gear stepper run current to 100% configured
20:52:56 MMU [T2] >>> [En] >>>>>>| [Ex] ... [Nz] 940.6mm (e:938.2mm)
20:52:56 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Loading Ext' OLD_ACTION='Loading'
20:52:56 - DEBUG: Loading filament into extruder
20:52:56 - - TRACE: _ensure_safe_extruder_temperature: current_temp=199.99, paused_extruder_temp=200.0, current_target_temp=200.0, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
20:52:56 - DEBUG: Loading last 60.0mm to the nozzle...
20:52:56 - - TRACE: _movequeues_wait_moves(toolhead=True, mmu_toolhead=True)
20:52:57 - - - STEPPER: GEAR+EXTRUDER: dist=60.0, speed=16.0, accel=400.0, sync=False, wait=True
20:52:57 - - TRACE: _movequeues_wait_moves(toolhead=True, mmu_toolhead=True)
20:53:01 - - TRACE: _movequeues_wait_moves(toolhead=True, mmu_toolhead=True)
20:53:01 - - TRACE: Loading filament to nozzle. Stepper: 'gear+extruder' moved 60.0mm, encoder measured 58.7mm (delta 1.3mm). Pos: @1000.6, (996.9mm)
20:53:01 - DEBUG: Total measured movement: 58.7mm, total delta: 1.3mm
20:53:01 - - TRACE: _movequeues_wait_moves(toolhead=True, mmu_toolhead=True)
20:53:01 MMU [T2] >>> [En] >>>>>>> [Ex] >>> [Nz] LOADED 1000.6mm (e:998.3mm)
20:53:01 - DEBUG: Filament should loaded to nozzle
20:53:01 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Loading' OLD_ACTION='Loading Ext'
20:53:01 - - TRACE: _movequeues_wait_moves(toolhead=True, mmu_toolhead=True)
20:53:01 Load of 1000.6mm filament successful (encoder measured 998.3mm)
20:53:01 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Loading'
```

moggieuk, 2023-11-19:
Hi.  WIth HH v2 the "logic" behind the load is slightly different but not a lot although extruder loading is not always synced.  The ability to run without home may be able to be forced and I may have inadvertently disabled that. but, the ERCF design cannot be truly reliable with some kind of homing point near to the extruder.  I would never run without a toolhead sensor but Happy Hare now supports pre-extruder sensor as an option as well as "touch" (stallguard) detection.  It's an art to turn the motor current and homing step size.

The problem with collision detection is that it's a hack.  I support it in Happy Hare because it it got traction but I've never liked it.  It can be made more reliable in some cases but increasing the homing step size.

On specific errors:
The log attached shows that the "homing (collision)" move of 2.1 was only seen as a single pulse (0.7mm) and this is enough to trigger the logic that the extruder has been hit.  Either the grip is poor or the encoder is misbehaving (btw the default encoder is prone to errors - one missed reflection at a critical point and it can tip the balance one way or the other - there really isn't any better way other than increasing homing step size which will increase the range the encoder reads over).  Again a pre-extruder sensor / endstop would fix this 100%

On the failure to load the bowden case.. it is true that it it doesn't error out any more but rather get warns.  The assumption is that the subsequent extruder homing would fail.  I can't tell you how many people shouted to make this a warning rather than an error because of their poor encoders.

moggieuk, 2023-11-19:
That said, I will check why `home_to_extruder: 0` doesn't turn off the homing step because it should...

BCRans, 2023-11-19:
Thank you for your reply, now I understand the logic and the reasons. Maybe as an final validation, the last 5-10mm which should be run purely from the extruder motor, and this could be validated by encoder to prove the filament is in the toolhead. This is already an option during unload, I would be very happy to see it also during the load process

And why am I running without a toolhead sensor - I have the original hall effect one, sometimes tiny hair gets stuck below extruder gears, it is enough to not let the little washer back into its place and I have to manually push a piece of filament to "clear it". And yes, I am planning to implement a proper microswitch based one. The issue is, I have a custom toolhead with absolutely no possibility to somehow implement that switch, I have to make a new design from scratch, and to iron out a couple other bugs with my design (weak part cooling for 10mm3+ flow rates, cable management, extruder just barely strong enough), for which I did not have time until now. 

Apart from this homing step, here are my other 2 possible issues - I use Orcaslicer now, when multicolor print finishes it makes a last partial filament change, ramming + cooling moves, and leaves the filament in toolhead, approx 30mm from the nozzle. In my print end macro I have the eject command, which starts the complete unload sequence, ramming again, cooling moves, and moves the filament into parking position by another 30mm and clears the extruder gears. Then it tries to extract the filament, but since the extruder is not gripping, encoder does not detect any movement and it pauses the end print macro. I think the "slicer park pos" parameter could help here and I adjusted it to "35" to match my slicer park position, but I think this parameter needs more explanation also in the docs - in which load/unload step it is used, and to which direction it offsets the filament position. I now have a couple single filament prints to do, I will return to test this after. 
Second one, when during pause I have to extrude filament due to whatever reason, with disconnected bowden the encoder does not sense movement, triggering a "clog", which tries to run the pause and park macro again. I am using the default ones, even with the conditionals it tries to park and pause it again. Is there a possibility to disable the clog detection during a PAUSE event?

Thanks a lot!

BCRans, 2023-11-24:
Update: 
tried again the old parameter of home_to_extruder=0, doesn't seem to work on my machine - tried both the mmu and ercf compatibility options. Anyway the Savox servo is sent, I think most of my issues will be solved just by that. During winter I think I will make a new toolhead design with a filament sensor (the hall effect didn't work for me reliably, the tiny hairs from hairy filament was enough to register as filament stuck).

Other little bits: 
the bug fix for the mmu config files include is working on my machine, thank you!
I had a strange error while paused due to my filament buffer getting stuck (working on a new design, in testing), disconnected the bowden and extruded filament, but since bowden was disconnected another "clog" was detected and the printer errored out. Is it possible to disable the clog detection during pause events? For now I included the mmu_test_config option in my macros to do that, but it spams my console.

closed 2024-12-23 by github-actions[bot]