moggieuk/Happy-Hare #89: Error after Update and reinstall

daTobi1, 2023-11-17:
Good day,
Since reinstalling happy hare after the update I get the following error:
"Klipper reports: ERROR
Option 'microsteps' in section 'extruder' must be specified
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is stopped"
I have defined the microsteps in my printer.cfg in the extruder section and can't find any instructions as to where they still need to be entered. Please help. Where do the microsteps have to be?

ningpj, 2023-11-17:
This happens in some configurations for some reasons. While its investigated, you can usually work around this by moving your extruder definition and loading it before HH config's. e.g. save your extruder section to a separate file and include it before other HH config.

LostFool, 2023-11-18:
Can confirm this works.

moggieuk, 2023-11-19:
All. The temporary workaround is to move the mmu includes to after the extruder, but I just pushed what I hope is a fix that will allow the mmu includes to be anywhere in the printer.cfg.  At the beginning is preferred.

closed 2024-02-05 by moggieuk