moggieuk/Happy-Hare #90: Installation on multiple instances fails on checking the service name

cicinovec, 2023-11-19:
install.sh script, line 95
Script is searching for a service called klipper.service
But on installations with multiple instances the service name is different, mostly klipper-[name].service

When changing the line, the installation proceeds with no other issues
See in attached file my file as example
[install.zip](https://github.com/moggieuk/Happy-Hare/files/13403008/install.zip)

NNGB-kor, 2023-12-07:
Installation also failed in multi-instance.
It tells me to install klipper first.

cicinovec, 2023-12-10:
Hi @moggieuk, with multiple instances you got `printer_data` renamed to `{whatevername}_data` (you can use `-c` and `-k`) but also the service name is changes to this format `klipper-{whatevername}.service`
The script should use `whatevername` as variable and use this variable to check klipper service availability.
This is as far as I can get with my programming knowledge. I can test it for you, but dont know how to update the script properly (in my installation I have changed the line directly to my service name `klipper-Voron24.service`)
[image]

yourbuddydinec, 2024-05-31:
I edited the current script to include a -n style command for printer name and it seems to install okay but get this error at the end. Is this "normal" or so I need to change things? 
Was able to install with `./install.sh -i -n E3 -z` with E3 being the name of my printer
[image]

closed 2024-12-14 by github-actions[bot]