moggieuk/Happy-Hare #92: MmuToolHead clobbers extruder configuration information

GarthSnyder, 2023-11-23:
I must apologize up front for not reporting this in combination with a PR, but after looking at the code, it's a bit beyond my Klipper-fu. I'm not sure what the proper resolution would be.

In `MmuToolHead.__init__`, the following entries are erased from the configuration database for `extruder`:

- `step_pin`
- `dir_pin`
- `enable_pin`
- `endstop_pin`
- `rotation_distance`
- `gear_ratio`,
- `microsteps`
- `full_steps_per_rotation`
- `pressure_advance`
- `pressure_advance_smooth_time`

A comment explains that this is done to `Nullify original extruder stepper definition so Klipper doesn't try to create it again`.

The problem is that this information may be relied upon by macros outside of Happy Hare. Furthermore, there doesn't seem to be any way to recover the information once nullification has occurred. In my case, I discovered the problem through the macro

```
[gcode_macro SET_DEFAULT_PRESSURE_ADVANCE]
gcode:
    {% set PA = printer.configfile.config.extruder.pressure_advance %}
    M118 Setting pressure advance to {PA}
    SET_PRESSURE_ADVANCE ADVANCE={PA}
```

which now causes the print to abort when Happy Hare is loaded. It's easy to prevent the aborting, not so easy to make the macro actually work again.

I'm tempted to say "Whatever, as long as there's some workaround I can use get the information." But really, HH shouldn't be breaking unrelated code or requiring that code to behave differently under HH. HH is interfering with Klipper's documented API.

If someone has a sense of what a better approach might be, I'd be happy to attempt the grunt work and submit a PR.

GarthSnyder, 2023-12-08:
Just to reiterate, I'm delighted with Happy Hare and appreciative of the (obviously extensive) work that's gone into it. If you can summarize an alternate approach in a few sentences, I could probably come up with a PR. I know you've got plenty to do already. 🙂

moggieuk, 2023-12-08:
No worries.  Not restoring was an oversight.  It's working in my development branch and I will add to upcoming release of v2.3.   As with any 2.x release it will need to re-run "./install.sh" because it has an additional klipper module.

It turns out is was very simple -- I just remember the nulled values and restore them on connect().  Nobody knows they were temporarily removed to stop Klipper creating an additional extruder object.  All good.

moggieuk, 2023-12-14:
Fixed in V2.3 update...

closed 2024-02-05 by moggieuk