moggieuk/Happy-Hare #93: updated to version: v2.2.0-5-gaa231409 --> Option 'step_pin' is not valid in section 'stepper_mmu_gear'

Eisbaeeer, 2023-11-23:
I reinstalled happy-hare from the srcatch after failure with the update by moving all [extruder] settings from hh to printer.cfg
Ít worked like a charme. Today next update and now get the following error:

`Option 'step_pin' is not valid in section 'stepper_mmu_gear'`

ningpj, 2023-11-23:
This unfortunately is a known issue....while its investigated, you can work around this by either moving your [include mmu/base/*.cfg] until after your [extruder] section or export your [extruder] section to a separate file and include it before these.

noolsr, 2023-12-03:
Hey moved my [extruder] section out of my printer config, included it before my [include mmu/base/*.cfg]. Still shows the same error. Any new updates on the bug?

moggieuk, 2023-12-08:
This shouldn't be a problem... sometimes I've seen it with incorrect pin_alias.  There used to be an issue with `[include]` order but as far as I'm aware that was fixed about a month ago with v2.2 update.

If, after upgrading to latest, you still have issues. Attach your `klippy.log`

Going to mark and not_a_bug / need_info

Riconec, 2023-12-10:
have the same error, first time installing HH

hfog, 2023-12-12:
Just upgraded Klipper (to deal with a different issue) and also HH and now I'm hitting this same error.
None of the suggestions elsewhere in this thread made any difference.

Riconec, 2023-12-12:
I installed HH and got a lot of issues, googling issues came up here someone told to update klipper so I did and the problems were still there. After of hour of strugles I decided to just start over and reinstall HH (after I updated klipper) and this time it worked without step_pin issue

noolsr, 2023-12-13:
Can confirm this works did the same and the error is gone.

iBral, 2024-04-23:
i have same error, when i update klipper

closed 2024-12-14 by github-actions[bot]