moggieuk/Happy-Hare #94: Feature Request : More summary information

LostFool, 2023-11-24:
I like getting the swap summary. It would be nice if we had the number of times slips, auto-recovery, and general errors happens. If it could be per tool as a final log that would be awesome.

This will help people narrow down where problems exist, if it's just certain gates/filaments. I have my machine running good with the auto-recover but I'd like to tune it better with summaries so I don't ignore that issues are happening.

moggieuk, 2023-12-08:
`mmu_stats detail=1`

Check out the manual / readme .. there is a LOT of status data that can be obtained.  Also `mmu_log` contains this info.

moggieuk, 2023-12-19:
I meant to say that `MMU_STATS DETAIL=1` also gives you what I think you want.

closed 2024-01-27 by LostFool