moggieuk/Happy-Hare #98: ercf 1.1 / easybrd / no toolhead senor:  unloading fails, claims filament is stuck in extruder, but filament has been unloaded above the gears.

sanderboer, 2023-12-05:
Hi,

I just updated to today's rev (dec. 5 23), installed a new cfg tree, did the initial calibrations and now every tool change fails on unloading the current filament:

`TRACE: _ensure_safe_extruder_temperature: current_temp=250.34, paused_extruder_temp=250.0, current_target_temp=250.0, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
17:02:54 - DEBUG: Unloading last 30.0mm to exist the extruder...
17:03:00 - - TRACE: Unloading extruder. Stepper: 'extruder' moved -30.0mm, encoder measured 0.0mm (delta 30.0mm). Pos: @-70.0, (39.9mm)
17:03:00 - DEBUG: Total measured movement: 0.0mm, total delta: 30.0mm
17:03:00 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Exiting Ext'
17:03:00 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Unloading'
17:03:00 An issue with the MMU has been detected whilst out of a print
         Reason: Unload sequence failed: Filament seems to be stuck in the extruder. Encoder not sensing any movement.
         Occured when changing tool: > T0
17:03:00 - - TRACE: Processing idle_timeout 'ready' event
`

I have poked in the dark changing several settings in mmu_parameters, most notably toolhead_extruder_to_nozzle and the unloading speeds an and none of these changes make a difference.
It looks to me that when unloading after forming a tip, the encoder measures no movement: 'Stepper: 'extruder' moved -30.0mm, encoder measured 0.0mm (delta 30.0mm)'. This is weird to me, as I tested the encoder while extruding and retracting filament, both movements add to the value returned by MMU_ENCODER. 
For instance, with filament loaded mmu_encoder returns X, after extruding 50mm it returns x+100, after retracting  50mm it returns X+100-90

Is there somewhere an assumption that retracting filament reverses the encoder count?
  

Any help, tips, tricks is greatly appreciated.

thx,
S

sanderboer, 2023-12-05:
note: I just noticed that when unloading the ERCF gear does not move, but the servo is down.

BCRans, 2023-12-07:
+1, sometimes happened to me also, but rarely. Theory in my case: 

Slicer parked filament higher, than HH expected it is, so during normal tip forming the filament was ejected. so the during the actual eject the extruder was not able to move the filament, encoder read 0

Since I set parameter slicer_tip_park_pos: 35 I did not have this issue. Some more explanation on this parameter would be helpful.

moggieuk, 2023-12-08:
You are likely unloading the filament in the TIP_FORMING macro or the TIP_CUTTING macro.  If you fully eject and then pass back control to Happy Hare it will (correctly) assume that the filament is stuck because, without a toolhead sensor to known better, it sees no movement.

The fix is to ensure your don't accidentally eject from the extruder - let Happy Hare do that.

sanderboer, 2023-12-08:
It is true that _MMU_FORM_TIP_STANDALONE yeets the filament out of the extruder, but I am at a loss on how to mitigate this. 

I don't have anything particular in place with regards to tip forming, in fact I have reinstalled klipper and happy_hare from scratch, only keeping mmu_vars.cfg
Setting  slicer_tip_park_pos: 35 has no effect.

These are the steps I take to get to this point:
- start printer and heat up hotend to 240deg
- issue T0 to load first filament
- issue T1 to switch to second filament:

`5:22 PM
An issue with the MMU has been detected whilst out of a print
Reason: Unload sequence failed: Filament seems to be stuck in the extruder. Encoder not sensing any movement.
Occured when changing tool: T0 > T2
5:22 PM
MMU [T0] <<< [En] <<<<<<< [Ex] <<. [Nz] -35.0mm (e:35.2mm)
5:22 PM
pressure_advance: 0.035000
pressure_advance_smooth_time: 0.040000
5:22 PM
pressure_advance: 0.000000
pressure_advance_smooth_time: 0.040000
5:22 PM
Forming tip...
5:22 PM
MMU [T0] <<< [En] <<<<<<< [Ex] <<< [Nz] LOADED 0.0mm (e:0.0mm)`

sanderboer, 2023-12-13:
I spent a little time poking in the dark with:

```
_MMU_FORM_TIP_STANDALONE 
COOLING_TUBE_LENGTH=10 
COOLING_TUBE_POSITION=30 
INITIAL_COOLING_SPEED=20 
FINAL_COOLING_SPEED=100 
COOLING_MOVES=4 
USE_SKINNYDIP=0 
USE_FAST_SKINNYDIP=0 
SKINNYDIP_DISTANCE=25 
DIP_INSERTION_SPEED=30 
DIP_EXTRACTION_SPEED=100 
MELT_ZONE_PAUSE=0 
COOLING_ZONE_PAUSE=1000 
UNLOADING_SPEED_START=100 
UNLOADING_SPEED=30 
RAMMING_VOLUME=10
 FINAL_EJECT=0 
SS_RAMMING=0
```

and I found that I cannot use skinnydip no matter the value of SKINNY_DIP_DISTANCE,

sanderboer, 2023-12-15:
Solved.
Turns out that when upgrading from an afterburrner to a stealthburner the new type of extruder motor not only has a different gear_ratio (50:10) but also the microsteps moves from the original 16 to 32. 
This last one has been part of the default settings since februari 2022, but I built my printer in jan. 2021 ( v 2.1184).

closed 2023-12-15 by sanderboer