moggieuk/Happy-Hare #99: 'MmuToolHead' object has no attribute 'need_flush_time'

matthewj301, 2023-12-08:
**When**
After updating 

Klipper version: v0.12.0-25-g99d7af87
Happy Hare: v2.2.0-7-g4819b69e

**Trace**
`
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 180, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/mmu.py", line 776, in handle_connect
    self._load_persisted_state()
  File "/home/pi/klipper/klippy/extras/mmu.py", line 861, in _load_persisted_state
    self._set_selector_pos(offset)
  File "/home/pi/klipper/klippy/extras/mmu.py", line 3650, in _set_selector_pos
    self.mmu_toolhead.set_position(pos, homing_axes=(0,))
  File "/home/pi/klipper/klippy/extras/mmu_toolhead.py", line 132, in set_position
    super(MmuToolHead, self).set_position(newpos, homing_axes)
  File "/home/pi/klipper/klippy/toolhead.py", line 438, in set_position
    self.flush_step_generation()
  File "/home/pi/klipper/klippy/toolhead.py", line 355, in flush_step_generation
    self._advance_flush_time(self.need_flush_time)
AttributeError: 'MmuToolHead' object has no attribute 'need_flush_time'
`

**Symptoms**
I started getting the error "'MmuToolHead' object has no attribute 'need_flush_time'". once ratOS started back up. 

**Investigation**
Looking at the above trace, it seems like Klipper is expecting the MmuToolHead object to have the 'need_flush_time' variable now.

Looking for that var in the Klipper codebase, it seems like a breaking change was just put in at this [commit ID](https://github.com/Klipper3d/klipper/commit/b7b13588c7bf8925e739954860eedc3a5eea42e7)

_Specifically_: Rename last_kin_move_time to need_flush_time and rename
force_flush_time to last_flush_time to improve variable name clarity.

I see last_kin_move_time, so this may be simply renaming those vars. Why they would make this a breaking change, rather than checking for both vars and issuing a warning to those who still use last_kin_move_time and force_flush_time, who knows :) 

I'll test the above and report back

matthewj301, 2023-12-08:
Btw, looks like there were more changes. Although my system starts up, it now complains about no 'priming_timer', seems releated to this [commit](https://github.com/Klipper3d/klipper/commit/9e574c34970fe4c68c5ad44d4f1f5faac4682e1e)

matthewj301, 2023-12-08:
Awesome, thanks for the fix and the additional info! 

I'll try out the update tomorrow, feel free to close my PR

moggieuk, 2023-12-08:
Thanks for your fix!  It needed a little more update to match Klipper... but I certainly appreciate the proactive diving into the issue.

closed 2024-02-05 by moggieuk