moggieuk/Happy-Hare #102: Internal error during connect: module 'toolhead' has no attribute 'BUFFER_TIME_HIGH'

GavinHanson, 2023-12-11:
Tried setting this to static `0` within the mmu_toolhead.py and it then allowed it to connect but i then get this message in the console:

```
15:13
Warning: Error booting up MMU: 'MmuToolHead' object has no attribute 'buffer_time_high'
15:13
Gates: |#0 |#1 |#2 |#3 |#4 |#5 |#6 |#7 |#8 |
Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |T8 |
Avail: | ? | ? | ? | ? | ? | ? | ? | ? | ? |
Selct: ------------------------------------- NOT HOMED
15:13
(\_/)
( *,*)
(")_(") Happy Hare Ready
15:13

```

sanderboer, 2023-12-12:
I am afraid you need to update your klipper install to the latest git revision.

moggieuk, 2023-12-14:
Klipper made a change that was incompatible with Happy Hare.  So you need to both upgrade Klipper and Happy Hare to get into a "happy" spot again.  Doesn't happen very often but it can...

closed 2024-02-05 by moggieuk