moggieuk/Happy-Hare #105: clog detection not always working

rwijnhov, 2023-12-14:
I am on the latest version of happy hare i use blinky. Sometimes if i filament doesn't load it wil just start printing with no filament loaded. The led of blinky doesn't blink but nog clog detection is triggered.  I see it give a loading error but it still starts printing. Clog detection is set on auto. I use a MMb board. setup like:
mmu_vendor: Other #ERCF			# MMU family
mmu_version: 1.1sb			# MMU hardware version number (add mod suffix documented above)
mmu_num_gates: 5 			# Number of selector gates
 
Alse tried setting Other to ERCF still same issue.

moggieuk, 2023-12-14:
Without the `mmu.log` it is impossible to help.  I'm confident this does work so could be a config error.   Have you reviewed the `mmu.log` yourself and compared against the doc.  The load and upload sequences are documented here: https://github.com/moggieuk/Happy-Hare?tab=readme-ov-file#---filament-loading-and-unloading-sequences

rwijnhov, 2023-12-16:
I upgraded to 2.3 no issues with clog detection till now.

moggieuk, 2024-03-11:
Closing because no follow up log file was provided