moggieuk/Happy-Hare #108: Implement support for EjectionAssistedPurge

mrmultifunk, 2023-12-14:
Would it be possible to implement support for purge units, like the one below?

[https://github.com/bombela/EjectionAssistedPurge](url)

I'm doing it manually with the _MMU_POST_UNLOAD macro in the moment, but i'm struggling with passing the slicer variables this way.

moggieuk, 2023-12-18:
At some point yes, this is on the enhancement list...

moggieuk, 2026-10-21:
Thanks for the request, and sorry for the long wait. Happy Hare v4 (v4.1 is the current release) now has a proper hook for purge devices like this, so I'm closing it.

Instead of hacking it into `_MMU_POST_LOAD`, you can give Happy Hare your own purge macro:
- In menuconfig, open "Purging" and set "Happy Hare purging method" to "Other / manually entered", then enter your macro name. This sets `purge_macro`.
- Turn on "Happy Hare controlled in-print purge" (`force_purge_standalone: 1`) if you want it to purge during prints too. Remember to turn off the slicer wipe tower.

The slicer data you were missing is now available to the macro:
- `printer.mmu.toolchange_purge_volume`: the volume to purge for this toolchange
- `printer.mmu.extruder_filament_remaining`
- `printer.mmu.slicer_tool_map.purge_volumes`: the full purge matrix

The shipped `_MMU_PURGE` macro (and the Blobifier macro) are good templates to copy.

If v4 is still missing something you need for EjectionAssistedPurge, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk