moggieuk/Happy-Hare #109: Mismatch in gantry servo naming

mrmultifunk, 2023-12-14:
In the _FILAMETRIX_GANTRY_SERVO_UP and _FILAMETRIX_GANTRY_SERVO_DOWN macros, the gantry servo is referenced as mmu_gantry_servo, but in the hardware definition it is named filametrix_gantry_servo.

Renaming the mmu_servo to mmu_gantry_servo made it all work.

moggieuk, 2023-12-18:
Oops. Thank you -- obviously I don't have a gantry servo yet. Fixed in next update.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

The gantry servo name in the config and the cut-tip macros now match in v4, so no renaming is needed.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk