moggieuk/Happy-Hare #110: I look forward to implementing your great work in Dog Matrix - AMO

LauOtero, 2023-12-15:
[image]

Great job on the new version, it makes my life easier for the new project I am developing.

It looks like it contemplates a design similar to Bambu Lab's AMS, may my design be based on that system.

Dog Matrix - AMO
[image]

My intention is to have 2 versions, normal and XL for 2kg filament spools.
It would be great to implement a solution for the Knomi V1 and V2 touch.

Thank you for your great work!!! 👏🫶

erazor666, 2023-12-17:
Looks cool! Is there any public info on this unit yet? Always looking for my next project 😎

moggieuk, 2023-12-18:
Very interesting.  So I would describe this as a "type-B" MMU.  See: https://github.com/moggieuk/Happy-Hare/blob/main/doc/conceptual_mmu.md

Type-B support is almost done.  In fact I was doing some work today on it...

erazor666, 2023-12-22:
@LauOtero this looks really cool and like a very worthwhile project, how far along is it? Any github?

LauOtero, 2023-12-29:
> @LauOteroEsto se ve realmente genial y es un proyecto que vale la pena, ¿qué tan avanzado está? ¿Algún github?

The design of certain components still needs to be finished in order to start testing and verify that everything works correctly.

One of the mods I'm designing is a dehumidifier, it complicates the design a bit trying to make it as airtight as possible.

There is private github but it is in Spanish, STEPS and BOM of some component, little documentation. I also manage with a partner a telegram channel dedicated to klipper in Spanish with a specific topic with the progress of the project.

Basically I'm doing the project by myself, when it's more advanced I hope to have more collaborators.

LauOtero, 2023-12-29:
Exactly, it is a type B MMU, in fact it is based on Bambu Lab's AMS.

It is designed to not perform calibrations, here (Spain) do not like much calibration with the encoder, I try to avoid any kind of calibration, I want it to be plug & play.

I adapt to what you program, I need to study the operation of the MMU in depth but I do not have the hours in the day to do it all.

Thanks for the great work you do.

> Very interesting. So I would describe this as a "type-B" MMU. See: https://github.com/moggieuk/Happy-Hare/blob/main/doc/conceptual_mmu.md
> 
> Type-B support is almost done. In fact I was doing some work today on it...

Exactly, it is a type B MMU, in fact it is based on Bambu Lab's AMS.

It is designed to not perform calibrations, here (Spain) do not like much calibration with the encoder, I try to avoid any kind of calibration, I want it to be plug & play.

I adapt to what you program, I need to study the operation of the MMU in depth but I do not have the hours in the day to do it all.

Thanks for the great work you do.

moggieuk, 2026-10-21:
Thanks for sharing this, it's a great looking design! Type-B MMUs like this (a separate gear motor per gate, no selector) are fully supported in Happy Hare v4 (v4.1 is the current release), so I'm closing this.

To set one up, pick "Custom Design" as the MMU Type in menuconfig and choose "VirtualSelector" as the selector type, which is the type-B option. Then add your sensors and options under "MMU Features / Additions". Several AMS-style units already ship as built-in types and are worth looking at as examples. If the Dog Matrix is released, I'm happy to look at adding it as a built-in type too.

If there's something your design needs that v4 doesn't do, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk